RPi-Distro/raspberrypi-sys-mods #1: echo statement not adequate in /etc/sysctl.d/98-rpi.conf

m-reiter, 2016-09-16:
The abovementioned file contains an echo statement which is apparently not adequate in a sysctl.conf file. Consequently, starting systemd-sysctl.service via "systemctl start" produces an error message. The second line in the file should probably read

vm.min_free_kbytes = 16384

On a further note, this settings conflicts with "vm.min_free_kbytes = 8192" found in /etc/sysctl.d/99-sysctl.conf under "rpi tweaks".

XECDesign, 2016-09-16:
That's quite bad. Thanks again.

XECDesign, 2016-09-16:
Would you happen to know what image you started with? rpi tweaks in 99-sysctl.conf seems to be from the wheezy days.

m-reiter, 2016-09-16:
I can't tell you exactly which image was used, but this is on a Pi 2 I got on October 9, 2015, and installed on October 9 or 10 with the then latest Jessie image from the foundation's site. If it's any help, I can check whether I still have the image file lying around.

I can confirm that on a Pi 3 installed in May this year with the then latest image, the rpi tweaks are not present in 99-sysctl.conf.

XECDesign, 2016-09-16:
would you be able to confirm that `sed -n '1h;1!H;${;g;s/\n\# rpi tweaks\nvm\.swappiness=1\nvm\.min_free_kbytes = 8192\n//g;p;}' /etc/sysctl.d/99-sysctl.conf` prints out the content of 99-sysctl.conf with the tweaks removed?

m-reiter, 2016-09-16:
Nope, this prints out the complete unchanged contents of 99-sysctl.conf

I'll attach the file for you so you can test things yourself.

[99-sysctl.txt](https://github.com/RPi-Distro/raspberrypi-sys-mods/files/477026/99-sysctl.txt)

XECDesign, 2016-09-16:
Thanks. Hopefully this should resolve the issue.
[raspberrypi-sys-mods_20160916_all.zip](https://github.com/RPi-Distro/raspberrypi-sys-mods/files/477114/raspberrypi-sys-mods_20160916_all.zip)

DNPNWO, 2016-09-16:
original file: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=160003&p=1038431#p1038431

DNPNWO, 2016-09-16:
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=160003&p=1038431#p1038431

m-reiter, 2016-09-16:
Unfortunately, if I install the .deb file, it does not remove the old tweaks. Calling the sed command from postinst manually does remove them, however.

XECDesign, 2016-09-16:
Did the "Removing old sysctl tweaks if present..." line show up?

DNPNWO, 2016-09-16:
I reverted back to the current Raspbian Image updated as of the 14th, How can I resolve the issue?

XECDesign, 2016-09-16:
@DNPNWO either wait for a fixed package to reach the repo, or install it manually:
```
wget https://github.com/RPi-Distro/raspberrypi-sys-mods/files/477114/raspberrypi-sys-mods_20160916_all.zip
unzip raspberrypi-sys-mods_20160916_all.zip 
sudo dpkg -i raspberrypi-sys-mods_20160916_all.deb
```
It looks like there may still be a much more minor issue remaining, but it shouldn't affect you.

DNPNWO, 2016-09-16:
Just out of curiosity, Does no one actually check to make sure the updates work prior to releasing it for the 'apt-get upgrade && apt-get update"?

XECDesign, 2016-09-16:
Clearly not thoroughly enough in this particular case (my fault), but yes, we check.

pelwell, 2016-09-16:
There's a balance between thoroughness and responsiveness, and sometimes things get overlooked (speaking from personal experience).

m-reiter, 2016-09-16:
Yes, it did.

DNPNWO, 2016-09-16:
Well, The I guess this is the last update/upgrade that I do for a while. 

I do not like spending hours of my time troubleshooting  bugs like this. Posting stuff like this on the foundation forum is an endeavor in futility. I posted about this issue yesterday, but got nowhere fast.  So I deleted that post, and spend hours troubleshooting it this morning, and reposted  my findings. 

This is ridiculous.

XECDesign, 2016-09-16:
@m-reiter Having trouble reproducing this one:
```
pi@raspberrypi:~ $ cat /etc/sysctl.d/99-sysctl.conf 
#
# /etc/sysctl.conf - Configuration file for setting system variables
# See /etc/sysctl.d/ for additional system variables.
# See sysctl.conf (5) for information.
#

#kernel.domainname = example.com

# Uncomment the following to stop low-level messages on console
kernel.printk = 3 4 1 3

##############################################################3
# Functions previously found in netbase
#

# Uncomment the next two lines to enable Spoof protection (reverse-path filter)
# Turn on Source Address Verification in all interfaces to
# prevent some spoofing attacks
#net.ipv4.conf.default.rp_filter=1
#net.ipv4.conf.all.rp_filter=1

# Uncomment the next line to enable TCP/IP SYN cookies
# See http://lwn.net/Articles/277146/
# Note: This may impact IPv6 TCP sessions too
#net.ipv4.tcp_syncookies=1

# Uncomment the next line to enable packet forwarding for IPv4
#net.ipv4.ip_forward=1

# Uncomment the next line to enable packet forwarding for IPv6
#  Enabling this option disables Stateless Address Autoconfiguration
#  based on Router Advertisements for this host
#net.ipv6.conf.all.forwarding=1

###################################################################
# Additional settings - these settings can improve the network
# security of the host and prevent against some network attacks
# including spoofing attacks and man in the middle attacks through
# redirection. Some network environments, however, require that these
# settings are disabled so review and enable them as needed.
#
# Do not accept ICMP redirects (prevent MITM attacks)
#net.ipv4.conf.all.accept_redirects = 0
#net.ipv6.conf.all.accept_redirects = 0
# _or_
# Accept ICMP redirects only for gateways listed in our default
# gateway list (enabled by default)
# net.ipv4.conf.all.secure_redirects = 1
#
# Do not send ICMP redirects (we are not a router)
#net.ipv4.conf.all.send_redirects = 0
#
# Do not accept IP source route packets (we are not a router)
#net.ipv4.conf.all.accept_source_route = 0
#net.ipv6.conf.all.accept_source_route = 0
#
# Log Martian Packets
#net.ipv4.conf.all.log_martians = 1
#

# rpi tweaks
vm.swappiness=1
vm.min_free_kbytes = 8192
pi@raspberrypi:~ $ sudo dpkg -i raspberrypi-sys-mods_20160916_all.deb 
(Reading database ... 66370 files and directories currently installed.)
Preparing to unpack raspberrypi-sys-mods_20160916_all.deb ...
Unpacking raspberrypi-sys-mods (20160916) over (20160915) ...
Setting up raspberrypi-sys-mods (20160916) ...
Installing new version of config file /etc/sysctl.d/98-rpi.conf ...
Removing old sysctl tweaks if present...
pi@raspberrypi:~ $ cat /etc/sysctl.d/99-sysctl.conf 
#
# /etc/sysctl.conf - Configuration file for setting system variables
# See /etc/sysctl.d/ for additional system variables.
# See sysctl.conf (5) for information.
#

#kernel.domainname = example.com

# Uncomment the following to stop low-level messages on console
kernel.printk = 3 4 1 3

##############################################################3
# Functions previously found in netbase
#

# Uncomment the next two lines to enable Spoof protection (reverse-path filter)
# Turn on Source Address Verification in all interfaces to
# prevent some spoofing attacks
#net.ipv4.conf.default.rp_filter=1
#net.ipv4.conf.all.rp_filter=1

# Uncomment the next line to enable TCP/IP SYN cookies
# See http://lwn.net/Articles/277146/
# Note: This may impact IPv6 TCP sessions too
#net.ipv4.tcp_syncookies=1

# Uncomment the next line to enable packet forwarding for IPv4
#net.ipv4.ip_forward=1

# Uncomment the next line to enable packet forwarding for IPv6
#  Enabling this option disables Stateless Address Autoconfiguration
#  based on Router Advertisements for this host
#net.ipv6.conf.all.forwarding=1

###################################################################
# Additional settings - these settings can improve the network
# security of the host and prevent against some network attacks
# including spoofing attacks and man in the middle attacks through
# redirection. Some network environments, however, require that these
# settings are disabled so review and enable them as needed.
#
# Do not accept ICMP redirects (prevent MITM attacks)
#net.ipv4.conf.all.accept_redirects = 0
#net.ipv6.conf.all.accept_redirects = 0
# _or_
# Accept ICMP redirects only for gateways listed in our default
# gateway list (enabled by default)
# net.ipv4.conf.all.secure_redirects = 1
#
# Do not send ICMP redirects (we are not a router)
#net.ipv4.conf.all.send_redirects = 0
#
# Do not accept IP source route packets (we are not a router)
#net.ipv4.conf.all.accept_source_route = 0
#net.ipv6.conf.all.accept_source_route = 0
#
# Log Martian Packets
#net.ipv4.conf.all.log_martians = 1
#

```
Appears to work as expected.

pelwell, 2016-09-16:
Possibly the reason this "error" wasn't noticed is that, apart from not changing the min_free_kbytes setting  and generating an error message, it has no ill effects. If you are going to get worked up about this then perhaps you should consider a different hobby?

m-reiter, 2016-09-16:
@XECDesign: Strange, I can reproduce it:

```
reiter@nacht:~ $ dpkg -s raspberrypi-sys-mods
Package: raspberrypi-sys-mods
Status: install ok installed
Priority: optional
Section: admin
Installed-Size: 62
Maintainer: Serge Schneider <serge@raspberrypi.org>
Architecture: armhf
Version: 20160915
Depends: init-system-helpers (>= 1.18~), libcap2-bin
Conffiles:
 /etc/sysctl.d/98-rpi.conf 12e2ecce9ec6fca160775acbd6099c17
 /etc/udev/rules.d/99-com.rules 279f8967ca1013b96aa59416fd8557ab
Description: System tweaks for the Raspberry Pi
 Various modifications to improve the performance or user experience.
Homepage: https://github.com/RPi-Distro/raspberrypi-sys-mods
reiter@nacht:~ $ cat /etc/sysctl.d/98-rpi.conf 
kernel.printk = 3 4 1 3
echo 16384 > /proc/sys/vm/min_free_kbytes
reiter@nacht:~ $ tail /etc/sysctl.conf
#net.ipv4.conf.all.accept_source_route = 0
#net.ipv6.conf.all.accept_source_route = 0
#
# Log Martian Packets
#net.ipv4.conf.all.log_martians = 1
#

# rpi tweaks
vm.swappiness=1
vm.min_free_kbytes = 8192
reiter@nacht:~ $ tail /etc/sysctl.conf|hexdump -C
00000000  23 6e 65 74 2e 69 70 76  34 2e 63 6f 6e 66 2e 61  |#net.ipv4.conf.a|
00000010  6c 6c 2e 61 63 63 65 70  74 5f 73 6f 75 72 63 65  |ll.accept_source|
00000020  5f 72 6f 75 74 65 20 3d  20 30 0a 23 6e 65 74 2e  |_route = 0.#net.|
00000030  69 70 76 36 2e 63 6f 6e  66 2e 61 6c 6c 2e 61 63  |ipv6.conf.all.ac|
00000040  63 65 70 74 5f 73 6f 75  72 63 65 5f 72 6f 75 74  |cept_source_rout|
00000050  65 20 3d 20 30 0a 23 0a  23 20 4c 6f 67 20 4d 61  |e = 0.#.# Log Ma|
00000060  72 74 69 61 6e 20 50 61  63 6b 65 74 73 0a 23 6e  |rtian Packets.#n|
00000070  65 74 2e 69 70 76 34 2e  63 6f 6e 66 2e 61 6c 6c  |et.ipv4.conf.all|
00000080  2e 6c 6f 67 5f 6d 61 72  74 69 61 6e 73 20 3d 20  |.log_martians = |
00000090  31 0a 23 0a 0a 23 20 72  70 69 20 74 77 65 61 6b  |1.#..# rpi tweak|
000000a0  73 0a 76 6d 2e 73 77 61  70 70 69 6e 65 73 73 3d  |s.vm.swappiness=|
000000b0  31 0a 76 6d 2e 6d 69 6e  5f 66 72 65 65 5f 6b 62  |1.vm.min_free_kb|
000000c0  79 74 65 73 20 3d 20 38  31 39 32 0a              |ytes = 8192.|
000000cc
reiter@nacht:~ $ sudo dpkg -i rpsm/raspberrypi-sys-mods_20160916_all.deb 
(Reading database ... 118967 files and directories currently installed.)
Preparing to unpack .../raspberrypi-sys-mods_20160916_all.deb ...
Unpacking raspberrypi-sys-mods (20160916) over (20160915) ...
Setting up raspberrypi-sys-mods (20160916) ...
Installing new version of config file /etc/sysctl.d/98-rpi.conf ...
Removing old sysctl tweaks if present...
reiter@nacht:~ $ cat /etc/sysctl.d/98-rpi.conf 
kernel.printk = 3 4 1 3
vm.min_free_kbytes = 16384
reiter@nacht:~ $ tail /etc/sysctl.conf
#net.ipv4.conf.all.accept_source_route = 0
#net.ipv6.conf.all.accept_source_route = 0
#
# Log Martian Packets
#net.ipv4.conf.all.log_martians = 1
#

# rpi tweaks
vm.swappiness=1
vm.min_free_kbytes = 8192
reiter@nacht:~ $ tail /etc/sysctl.conf|hexdump -C
00000000  23 6e 65 74 2e 69 70 76  34 2e 63 6f 6e 66 2e 61  |#net.ipv4.conf.a|
00000010  6c 6c 2e 61 63 63 65 70  74 5f 73 6f 75 72 63 65  |ll.accept_source|
00000020  5f 72 6f 75 74 65 20 3d  20 30 0a 23 6e 65 74 2e  |_route = 0.#net.|
00000030  69 70 76 36 2e 63 6f 6e  66 2e 61 6c 6c 2e 61 63  |ipv6.conf.all.ac|
00000040  63 65 70 74 5f 73 6f 75  72 63 65 5f 72 6f 75 74  |cept_source_rout|
00000050  65 20 3d 20 30 0a 23 0a  23 20 4c 6f 67 20 4d 61  |e = 0.#.# Log Ma|
00000060  72 74 69 61 6e 20 50 61  63 6b 65 74 73 0a 23 6e  |rtian Packets.#n|
00000070  65 74 2e 69 70 76 34 2e  63 6f 6e 66 2e 61 6c 6c  |et.ipv4.conf.all|
00000080  2e 6c 6f 67 5f 6d 61 72  74 69 61 6e 73 20 3d 20  |.log_martians = |
00000090  31 0a 23 0a 0a 23 20 72  70 69 20 74 77 65 61 6b  |1.#..# rpi tweak|
000000a0  73 0a 76 6d 2e 73 77 61  70 70 69 6e 65 73 73 3d  |s.vm.swappiness=|
000000b0  31 0a 76 6d 2e 6d 69 6e  5f 66 72 65 65 5f 6b 62  |1.vm.min_free_kb|
000000c0  79 74 65 73 20 3d 20 38  31 39 32 0a              |ytes = 8192.|
000000cc
```

DNPNWO, 2016-09-16:
> Possibly the reason this "error" wasn't noticed is that, apart from not changing the min_free_kbytes setting and generating an error message, it has no ill effects.

Really?  My guess is that the "raspberrypi-sys-mods" upgrade broke... and I noticed the error soon as I rebooted, clear as day on the boot screen. "[Failed] failed to start apply kernel variables"... regarding virtual memory (If I am not mistaken).

How can you miss that?

XECDesign, 2016-09-16:
@m-reiter did you by any change copy 99-sysctl.conf over sysctl.conf? Your original report was about 99-sysctl.conf, but the output you've provided is for sysctl.conf.

m-reiter, 2016-09-16:
Strange. I was sure I had seen before that 99-sysctl.conf was a symlink to /etc/sysctl.conf. Currently, that's no longer the case. However, apart from the rpi tweaks which are indeed removed in 99-sysctl.conf, both files are identical. Does this make any sense?

DNPNWO, 2016-09-16:
> @DNPNWO either wait for a fixed package to reach the repo, or install it manually

>It looks like there may still be a much more minor issue remaining, but it shouldn't affect you.

```
wget https://github.com/RPi-Distro/raspberrypi-sys-mods/files/477114/raspberrypi-sys-mods_20160916_all.zip
unzip raspberrypi-sys-mods_20160916_all.zip 
sudo dpkg -i raspberrypi-sys-mods_20160916_all.deb
```

If "raspberrypi-sys-mods" is still broken/unfixed I will not at this time install it at all, or even update it. I already restored my backup image, and I am on the 'raspberrypi-sys-mods' prior version from last week.

XECDesign, 2016-09-16:
@m-reiter good call:
```
pi@raspberrypi:~ $ ls -la test*
-rw-r--r-- 1 pi pi 5 Sep 16 16:32 test_file
lrwxrwxrwx 1 pi pi 9 Sep 16 16:32 test_symlink -> test_file
pi@raspberrypi:~ $ sed -n 's/test/fail/' -i test_symlink
pi@raspberrypi:~ $ ls -la test*
-rw-r--r-- 1 pi pi 5 Sep 16 16:32 test_file
-rw-r--r-- 1 pi pi 0 Sep 16 16:33 test_symlink
```
The right thing to do then is to run sed on sysctl.conf.

XECDesign, 2016-09-16:
Hopefully the final fixed package:
[raspberrypi-sys-mods_20160916_all.zip](https://github.com/RPi-Distro/raspberrypi-sys-mods/files/477318/raspberrypi-sys-mods_20160916_all.zip)

m-reiter, 2016-09-16:
Yes, this seems indeed to fix everything.

Just to be sure: 99-sysctl.conf should indeed be a symlink? Does this look right

```
reiter@nacht:~$ ls -l /etc/sysctl.d/99-sysctl.conf
lrwxrwxrwx 1 root root 14 Sep 16 19:18 /etc/sysctl.d/99-sysctl.conf -> ../sysctl.conf
```

or should it be absolute?

XECDesign, 2016-09-16:
It should be a symlink, but I don't remember whether the default is absolute or relative. The end result is the same in either case.

XECDesign, 2016-09-16:
From a 'clean' pi:
`lrwxrwxrwx   1 root root    14 Apr  2 20:33 99-sysctl.conf -> ../sysctl.conf`

giana77, 2016-09-16:
Hello, after today upgrade to new version of raspberrypi-sys-mods and reboot I'm getting huge fps drops from motion detection camera connected to RPi 3 even I did upgrade to raspberrypi-sys-mods_20160916_all.zip from this thread with dpkg. 
The second RPi 3 with camera and without upgrade works fine. Any suggestions please? Thanks.

XECDesign, 2016-09-17:
@giana77 That's not really related to this particular issue. I've opened a new one here https://github.com/RPi-Distro/raspberrypi-sys-mods/issues/2

closed 2016-09-17 by XECDesign

m-reiter, 2016-09-17:
@XECDesign Thanks for looking up the "clean" link - I've seen too many things make a difference that shouldn't have to not at least be wary in these things...