RPi-Distro/raspberrypi-sys-mods #4: No README and repository description

AlekSi, 2016-10-12:
It's hard to understand what it is without a README.

XECDesign, 2016-10-12:
It's hard to describe what this package is for other than to say it's for raspberry pi system mods... which is already in the title. Suggestions?

AlekSi, 2016-10-12:
Well, to describe what is system mods?
Short README is better than nothing. If it's something I'm not looking for, I can read README and move to next repository. Otherwise, I have to figure it out from sources.

pelwell, 2016-10-12:
"A collection of Raspberry Pi-sourced system configuration files and associated scripts"?

XECDesign, 2016-10-12:
Added that as the description. Thanks.

AlekSi, 2016-10-12:
It probably would be useful to describe (or add a link to description) its place in the whole build process.
Btw, first Google hit for "raspberry pi system mods": http://www.techrepublic.com/pictures/raspberry-pi-10-fun-and-fantastic-mods/. 😄

closed 2016-12-09 by XECDesign