RPi-Distro/raspberrypi-sys-mods #5: Postinst script syntax error

djpenny, 2016-10-18:
root@raspberrypi:/home/pi# apt-get dist-upgrade
Reading package lists... Done
Building dependency tree        
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up raspberrypi-sys-mods (20161018+1) ...
/var/lib/dpkg/info/raspberrypi-sys-mods.postinst: 49: /var/lib/dpkg/info/raspberrypi-sys-mods.postinst: Syntax error: ";;" unexpected (expecting "fi")
dpkg: error processing package raspberrypi-sys-mods (--configure):
 subprocess installed post-installation script returned error exit status 2
Errors were encountered while processing:
 raspberrypi-sys-mods
E: Sub-process /usr/bin/dpkg returned an error code (1)

XECDesign, 2016-10-18:
On it. Thanks.

djpenny, 2016-10-18:
Add 'fi' before line 45 :)

XECDesign, 2016-10-18:
Maybe one day I'll update this package without breaking anything.
Thanks again. Update should show up in the repo soon.

XECDesign, 2016-10-18:
Give it a go now.

bikedude880, 2016-10-18:
Works now, thanks for the speedy fix!

closed 2016-10-18 by XECDesign

bert269, 2022-01-09:
I just did an update on Bullseye for Raspberry Pi and it seems this problem has returned.
Any ideas?
Setting up raspberrypi-sys-mods (20220106) ...
Removing vscode repo stubs...
rm: cannot remove '/etc/apt/trusted.gpg.d/microsoft.gpg': Operation not permitte     d
dpkg: error processing package raspberrypi-sys-mods (--configure):
 installed raspberrypi-sys-mods package post-installation script subprocess retu     rned error exit status 1
Setting up raspi-config (20220106) ...
Errors were encountered while processing:
 raspberrypi-sys-mods
E: Sub-process /usr/bin/dpkg returned an error code (1)

Thanks

MichaIng, 2022-01-09:
> Operation not permitted

I wasn't able to replicate 🤔. Can you show the output of the following command:
```sh
ls -l /etc/apt/trusted.gpg.d/microsoft.gpg
```

However, you can solve it by removing it manually:
```sh
sudo rm /etc/apt/trusted.gpg.d/microsoft.gpg
```

XECDesign, 2022-01-10:
Doesn't look related to the initial issue.

IIRC, some people were telling others to run `chattr +i` on those files. I suspect a lot of people copied and pasted those instructions without understanding that it wouldn't help with their concerns and that it could lead to issues like this.

Can't think of what else might cause this.

I'll add a `||:` for it to continue anyway.

MichaIng, 2022-01-10:
> `chattr +i`

Ah yes that will be it, I remember this was suggested by some, even that simple `rm` did permanently remove the repo at any point in time.