RPi-Distro/raspberrypi-sys-mods #9: sshpasswd.sh triggers security notifications if user doesn't have sudo permissions

m3phisto, 2016-12-09:
Problem:
The sshpasswd.sh script in etc/profile.d performs the check even if the current user is not able to perform the sudo action required. This triggers a security notice (and in my case an email).

Mitigation:
A possible solution would be to check the "$USERNAME" variable for equality with "pi"/"root" before calling "check_hash".
I can prepare a Pull Request if necessary.

XECDesign, 2016-12-09:
See #8

spl237, 2016-12-09:
Checking $USERNAME won't help if the pi user has disabled passwordless sudo. We're working on a proper solution for this - bear with us.

closed 2016-12-09 by spl237

m3phisto, 2016-12-09:
Sorry for reporting a known issue. I didn't check closed issues (as this problem clearly persists).