RPi-Distro/raspberrypi-sys-mods #10: sshpasswd.sh makes mkpasswd complain about wrong salt lenth

fulmeek, 2016-12-11:
Hello,

On every login, mkpasswd complains about wrong salt length:
> Wrong salt length: 0 byte when 8 <= n <= 16 expected.

This is caused by the new sshpasswd.sh in /etc/pofile.d/ on line 10:
> HASH=$(mkpasswd -msha-512 raspberry "$SALT")

I think there's something off with the sed line that results in an empty $SALT variable. At least, you should check $SALT for emptyness before passing it to mkpasswd.

spl237, 2016-12-11:
I've no idea what's going wrong there - the previous tests in the file should ensure that there is a line with a valid password entry in the shadow file, so the grep for the salt shouldn't fail - the only thing likely to make it do so is if you are using non-standard encryption for your password file, and in that case most of the rest of this routine will fail anyway.

Could you please post the output of sudo grep -E '^pi:' /etc/shadow  - if you are concerned about security, delete the second half of the line; I only need to see the first 20 or so characters.

fulmeek, 2016-12-11:
Well, it's not my machine, so I can't provide the output you requested. The only thing I can tell is that
> sudo grep -E '^pi:' /etc/shadow

returns the line of the pi user, but
> SALT=$(echo "${SHADOW}" | sed -n 's/pi:\$6\$//;s/\$.*//p')

returns an empty string.

I'm not sure what exactly the regex pattern does, but if it looks for a ";" char, it's not in the string. The rest appears to match the pattern. It's a valid SHA-512 hash.

spl237, 2016-12-11:
No, the regex isn't looking for a semicolon - that just separates two different sed patterns. That sed should extract all the characters between the initial "pi:$6$" and the next "$" - the only way it will return a null string is if the line from the shadow file does not start "pi:$6$" and contain a third $ at some point.

The format of a line in the shadow file is <user>:$<encryption digit>$<salt>$<hash>

If you have a pi user, there will be a line starting "pi:$", so either the encryption used is not SHA-512, indicated by the digit not being a 6, or the following $ separator is missing.

Without the actual line in question, I can't really debug this. All I can say is that it hasn't been reported by anyone else, so the machine in question has probably had something unusual done to its password file, but I can't diagnose what it is by guesswork.

fulmeek, 2016-12-11:
Well, the line starts with:
> pi:$6$xxxxxxxx$xxx...

I hope you understand that I can't provide the actual string, especially because it's not my own machine. 

For your debugging, if I put that string into the command line, it's empty too:
> echo $(echo "pi:$6$xxxxxxxx$xxx..." | sed -n 's/pi:$6$//;s/$.*//p')

Hope you might be able to figure out what the issue could be.

spl237, 2016-12-11:
If echoing the string manually from the command line, you need to escape the $ symbols, or the shell mangles them. Try

`echo $(echo "pi:\$6\$xxxxxxxx\$xxx..." | sed -n 's/pi:\$6\$//;s/\$.*//p')`

instead. That works fine on my Pi, and produces the correct result.

It really doesn't compromise the security of the password file if you only share the first few characters of the hash - can you please just post the line up to a few characters after the third $ sign? For example, here's the equivalent from my file:

pi:$6$k0vFWRoG$bEB...

That is all I need to diagnose the problem; it doesn't reveal anything helpful to a hacker.

I did wonder if it was running in a strange shell which was causing unusual behaviour, but I've just tested on both sh and bash, and it does the same on them here.

spl237, 2016-12-11:
Actually, looking at it - are the $ signs escaped in the sed line in /etc/profile.d/sshpasswd.sh? They weren't when you quoted it.

So is the line in /etc/profile.d/sshpasswd,sh

`SALT=$(echo "${SHADOW}" | sed -n 's/pi:$6$//;s/$.*//p')`

as you quoted it, or is it actually

`SALT=$(echo "${SHADOW}" | sed -n 's/pi:\$6\$//;s/\$.*//p')`

which is what it should be?

It won't work if the slashes before the $ signs are missing - they should be there in the file as installed, and they are on my test machine. Has something / someone edited the file after install?

spl237, 2016-12-11:
SALT=$(echo "${SHADOW}" | sed -n 's/pi:\$6\$//;s/\$.*//p')

spl237, 2016-12-11:
Ah - I wonder if you have the characters "\n" or "\t" in the salt (the part of the line between the $6 and the second $. Is there a "\" character in the salt, and if so, what character follows it?

pelwell, 2016-12-11:
How about replacing all of the alphanumericss (a-z, A-Z, 0-9) with "x"s then posting the result? That should be safe.

spl237, 2016-12-11:
OK, as far as I can tell, the characters mkpasswd uses in creating a salt, or allows the use of in a salt, should all work fine in these scripts. So I really need to know what the salt you used was (obfuscated as Phil suggests above if it makes you more comfortable), or I can't make any further progress on this.

fulmeek, 2016-12-11:
Sorry that I can't help you much at this point, I only had access today because of this issue. It could be any character that makes it fail. On my own Pi, it works as expected.

Maybe you should check for an empty salt, just to be safe. If someone uses a different algorithm, it will fail for sure.

spl237, 2016-12-11:
No-one should be using any algorithm other than SHA-512 - that's the default for Raspbian and all the other options are less secure.

Closing until I can get more information, or someone can repeat the issue.

closed 2016-12-11 by spl237

pelwell, 2016-12-11:
Any password entry we can't parse is a changed password, so we shouldn't warn about it. If people change the password back to raspberry but with a different salt or encryption scheme then they are opting out of the warning.

spl237, 2016-12-11:
At present, the warning won't be generated if the salt is null anyway.

fulmeek, 2016-12-11:
Well, mkpassword complains because the salt is empty, and that's because the pattern didn't match. See the original post. It took a while to realize where this message came from.

spl237, 2016-12-11:
The _warning that the system is vulnerable_ will not be generated if the salt is null.