RPi-Distro/raspberrypi-sys-mods #11: touchscreen goes black on execution of package

RogueM, 2016-12-18:
For the last few drops I have noticed that each time `raspberrypi-sys-mods` gets an upgrade as part of a system `apt-get upgrade`, the RPF touchscreen goes black. I have also experienced that over HDMI feeding up a lapdock.

Today I went back to an earlier Raspbian image just to test the theory that `raspberrypi-sys-mod` was the culprit, and confirmed it is. In fact I can produce the problem on demand like so:

> sudo apt-get remove raspberrypi-sys-mod
> sudo apt-get install raspberrypi-sys-mod

spl237, 2016-12-18:
Do you mean it goes black temporarily during the update and recovers at the end, or that it goes black and stays back until you restart, or goes black and never recovers?

RogueM, 2016-12-18:
no, it never recovers. I need to reboot it over ssh (which is potentially dangerous if there is a kernel swap going on but hey!).

I'm also investigating this issue, which I have a hunch is related:
https://www.raspberrypi.org/forums/viewtopic.php?f=108&t=120968&p=1083463&hilit=sleep#p1083463

... my current theory is that when the system goes into sleep the service kicks in and shuts down the backlight. This is pure speculation however, but I'm going to try to kill that service/facility for a while and see if I still have problem waking up the touchscreen from sleep...

spl237, 2016-12-18:
OK, so when you say it never recovers, you mean it only recovers when you reboot it?

RogueM, 2016-12-18:
yes, correct. I can do any amount of work over ssh when the problem occurs (whether those 2 problems are related, which I have not gathered evidence yet that they are).

spl237, 2016-12-18:
OK, we'll investigate. Probably won't be until after Christmas now though! Thanks for the report.

RogueM, 2016-12-18:
np... I'll post here if I find any additional info worth relating.

RogueM, 2016-12-18:
right, looks like I can trigger the issue by doing:

`sudo systemctl restart rpi-display-backlight`

... which I suspect occurs as part of the package postinst?

spl237, 2016-12-18:
Not directly, but something else may be triggering it. The right answer is probably that restarting that service should work properly - I'll flag it to the relevant people.

commit 826b0232 (2016-12-20): Don't start or restart rpi-display-backlight

Fix typo in debian/rules

Fixes #11
closed 2016-12-20 by XECDesign