RPi-Distro/raspberrypi-sys-mods #12: sshpasswd.sh doesn't support custom number of rounds in crypted password string

ryan-greer, 2016-12-19:
The sshpasswd.sh script does not handle the salt correctly if the password string for the pi user contains `rounds=...` in the hashed password string.

Use case:

In this case I am configuring the pi user with Ansible and using a crypted password generated using passlib for python as described in this doc <http://docs.ansible.com/ansible/faq.html#how-do-i-generate-crypted-passwords-for-the-user-module>

This results in the following line in /etc/shadow: `pi:$6$rounds=656000$salt$password...`

Since sshpassword.sh does not expect this, it returns 'rounds=65600' as the SALT on line 9 and displays the following error when executing line 10:
```
Illegal salt character '='.
```

The above error appears to cause HASH to be an empty string which exits the script at line 11

Suggested solution:

Something along the lines of this should fix the issue: 
```
if echo $SHADOW | grep -q "pi:\$6\$rounds\="; then
     SALT=$(echo "${SHADOW}" | sed -n 's/pi:\$6\$rounds\=[0-9]*\$//;s/\$.*//p')
else
    SALT=$(echo "${SHADOW}" | sed -n 's/pi:\$6\$//;s/\$.*//p')
fi
```

spl237, 2016-12-19:
Thanks for reporting that - I'd never seen rounds used in a password file. The solution above won't work correctly, as it ignores the number of rounds when subsequently calculating the hash, but it's probably safe to assume that if someone has implemented rounds, they will also have changed the password, so we can just make it an early fall-out from the function. We'll include a fix in the next round of updates.

ryan-greer, 2016-12-19:
👍

idcrook, 2017-01-28:
I encountered this too.  it's part of SHA-512 crypt, it seems. And ansible is being explicit, as `rounds=65600` is the default from what I gather. 

The following also works to get a properly formed salt to compare against:
```
   SALT=$(echo "${SHADOW}" | sed -n 's/pi:\$6\$//; s/rounds=\d+\$//; s/\$.*//p')
```

I noticed that `/etc/profile.d/sshpasswd.sh` was removed from `master` branch in this repo?  Not sure why, as it seems like a helpful reminder/check.

idcrook, 2017-01-28:
Ahh, I see in (closed) PR #17 it was clarified that `/etc/profile.d/sshpasswd.sh` will be getting replaced by a PAM module.

XECDesign, 2017-01-28:
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=172787

XECDesign, 2017-03-05:
@spl237 Has this been fixed?

spl237, 2017-03-05:
Yes, the new PAM module should support all valid password encodings, including rounds

closed 2017-03-05 by spl237