RPi-Distro/raspberrypi-sys-mods #14: sshpasswd.sh fails if /usr/sbin is not in $PATH

mutability, 2017-01-17:
```
$ bash -x -l
[...]
+ for i in '/etc/profile.d/*.sh'
+ '[' -r /etc/profile.d/sshpasswd.sh ']'
+ . /etc/profile.d/sshpasswd.sh
++ service ssh status
bash: service: command not found
++ grep -q running
++ unset check_hash
```

spl237, 2017-01-17:
Very little we can do about that - /usr/sbin is in the path on a default Raspbian install, and we need to be able to access "service" to determine whether or not SSH is running. I fear that the easiest fix for this is either to put /usr/sbin back in the path, or to put "service" in a directory which is on the path.

closed 2017-01-17 by spl237

mutability, 2017-01-17:
Uh, the obvious fix here is to invoke `service` with an explicit path.

mutability, 2017-01-17:
Note that this is another case of an upgrade regression - existing installs, when upgraded, suddenly start seeing error messages on login.