RPi-Distro/raspberrypi-sys-mods #16: postinst appends to config.txt without an [all] section header

mutability, 2017-01-18:
raspberrypi-sys-mods postinst appends audio settings to /boot/config.txt. If the existing config.txt uses hardware-specific sections (`[pi3]`, etc) then those changes will only be applied to whatever hardware section was last active, not globally.

It should probably append an `[all]` before the audio changes.

XECDesign, 2017-01-21:
I see your point, but this is probably not going to happen. The users affected by this are those which have an image that's over a year, have heavily customised their config.txt, but haven't added the audio line, and are upgrading now. Feels like verybody who was going to hit this problem, would've already hit it and fixed up their config.txt appropriately.

mutability, 2017-01-21:
I don't understand; why would you not want to fix this bug? Does it break something else?

Note that it affects any image built by raspbian-ua-netinst (including images built right now)

mutability, 2017-01-21:
An alternative would be to change the postinst version test to lt-nl, not lt, so that fresh installs of the package do not touch config.txt at all (under the assumption that new installs already have a correct config.txt)

(That approach should really be applied to most of the postinst changes, tbh)

XECDesign, 2017-01-21:
I see no reason not to fix it, especially given that you've pointed out that it runs on upgrade AND install. I think we're on the same page now.

XECDesign, 2017-01-23:
To avoid making potentially unwanted changes, it will now avoid modifying config.txt files containing filters.

The reason for running those updated on install and upgrade rather than upgrade only is related to the original purpose of the package. Occasionally, we needed to modify files which were either not owned by any of our packages or owned by other packages. This package needed to be picked up on existing installs and make the appropriate changes. using the `-nl` alternative would break that.

closed 2017-01-23 by XECDesign

mutability, 2017-01-23:
Looks good, thanks!

commit b623fa2b (2017-01-23): check_audio: don't modify config.txt files with filters

Fixes #16