RPi-Distro/raspberrypi-sys-mods #18: ttyAMA0 -> Serial0 with kgdboc

macmpi, 2017-05-19:
Thanks for this very useful package.

I noticed that `cmdline.txt` [serial interface replacements](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/master/debian/raspberrypi-sys-mods.postinst#L44
) may not catch situations where both `console=ttyAMA0 `and `kgdboc=ttyAMA0` exist.
Is it intended?
Should only the first one be replaced, and not `kgdboc`?
Thanks for your thoughts.

PS: also unsure why `whois` dependency is required

pelwell, 2017-05-19:
The firmware is intentionally replacing `=ttyAMA0` with `=ttyS0` in situations where ttyS0 is the primary (console) UART. `kgbdoc` _could_ be handled seperately, but given that only one UART at a time is accessible on the 40-pin this is unlikely to be useful.

N.B. If you really want to use both UARTs on the command line, use the `serial0` (primary) and `serial1` (secondary) aliases instead - these are always replaced by the appropriate UART.

@XECDesign may be able to answer the `whois` question.

XECDesign, 2017-05-19:
`whois` was added because a script relied on `mkpasswd`. However, that has been replaced with https://github.com/raspberrypi-ui/pam/blob/master/modules/pam_chksshpwd/pam_chksshpwd.c and is no longer needed.

macmpi, 2017-05-19:
Thanks a lot for the quick responses!
I see dependency has already been removed: great!

closed 2017-05-19 by macmpi