RPi-Distro/raspberrypi-sys-mods #20: '/usr/lib/raspberrypi-sys-mods/rfkill-persist' failed with exit code 1.

atuyosi, 2017-12-23:
I found some error message about rfkill-persist. I don't know what this program originally intends.

It seems that saving state of "RF Kill flag" does not work,  and the wireless LAN device is invalidated each time system rebooted.

Error log are below:

```
$ LANG=C journalctl -b  | grep rfkill
Dec 23 17:10:05 raspberrypi systemd[1]: Listening on Load/Save RF Kill Switch Status /dev/rfkill Watch.
Dec 23 17:10:07 raspberrypi systemd-udevd[464]: Process '/usr/lib/raspberrypi-sys-mods/rfkill-persist' failed with exit code 1.
Dec 23 17:10:11 raspberrypi systemd-udevd[494]: Process '/usr/lib/raspberrypi-sys-mods/rfkill-persist' failed with exit code 1.
Dec 23 17:10:11 raspberrypi systemd-udevd[494]: Process '/usr/lib/raspberrypi-sys-mods/rfkill-persist' failed with exit code 1.
```

I checked script's source. It invoke `systemd-rfkill` as follows.

```
$ cat  /usr/lib/raspberrypi-sys-mods/rfkill-persist
#!/bin/sh
RFKID=`basename $DEVPATH`
/lib/systemd/systemd-rfkill save $RFKID
```

However, `systemd-rfkill` command isn't require any arguments.

```
 $ sudo /lib/systemd/systemd-rfkill save
This program requires no arguments.
```

Version information:

- Raspberry Pi 3B
- Raspbian Stretch 2017/11/29
- raspberrypi-sys-mods             20171127

```
$ uname -a
Linux raspberrypi 4.9.70-v7+ #1068 SMP Mon Dec 18 22:12:55 GMT 2017 armv7l GNU/Linux
```

pelwell, 2018-01-02:
It looks as though this service is no longer necessary using Stretch. Raspbian Jessie used systemd 215. As of systemd 227 the standard rfkill support automatically saves and restores the rfkill state with each change. Raspbian Stretch is now using systemd 232, so we can remove the `rfkill-persist` script and the corresponding `udev` rule.

XECDesign, 2018-01-03:
An updates sys-mods package has been pushed out. Please close if it resolves the issue.

atuyosi, 2018-01-03:
Thank you for quick fix.

No error message  reported by systems-udevd. 

I did `rfkill unblock wifi' , and  I confirmed that the flag was changed.  
But, the RF flag still remains to be "blocked" each time the system is restarted.

```
$ cat /var/lib/systemd/rfkill/platform-3f300000.mmc\:wlan
1
$ rfkill unblock wifi
$ cat /var/lib/systemd/rfkill/platform-3f300000.mmc\:wlan
0
$ rfkill list
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
```

Is the initial value specified somewhere?

```
$ LANG=C journalctl -b | egrep -E 'rfkill|RF'
Jan 04 02:13:33 stretch-lite systemd[1]: Listening on Load/Save RF Kill Switch Status /dev/rfkill Watch.
Jan 04 02:13:33 stretch-lite systemd[1]: Starting Load/Save RF Kill Switch Status...
Jan 04 02:13:33 stretch-lite systemd[1]: Started Load/Save RF Kill Switch Status.
Jan 04 02:13:36 stretch-lite dhcpcd[354]: wlan0: if_up: Operation not possible due to RF-kill
Jan 04 02:13:40 stretch-lite kernel: Bluetooth: RFCOMM TTY layer initialized
Jan 04 02:13:40 stretch-lite kernel: Bluetooth: RFCOMM socket layer initialized
Jan 04 02:13:40 stretch-lite kernel: Bluetooth: RFCOMM ver 1.11
Jan 04 02:24:09 stretch-lite systemd[1]: Starting Load/Save RF Kill Switch Status...
Jan 04 02:24:09 stretch-lite systemd[1]: Started Load/Save RF Kill Switch Status.
```

XECDesign, 2018-03-23:
@atuyosi Any luck with this? I haven't managed to reproduce the problem you're seeing.

atuyosi, 2018-03-24:
@XECDesign 

I confirmed that this problem does not occur in the latest version Raspbian and latest firmware.

I think this issue solved. Thank you.

cyberatz, 2018-03-25:
still having this issue. updated and rebooted to the latest Raspbian release.

XECDesign, 2018-03-25:
What's the output of these commands?
```
apt-cache policy raspberrypi-sys-mods
sudo systemctl status rfkill-persist
```

closed 2018-04-06 by XECDesign