RPi-Distro/raspberrypi-sys-mods #21: new version of raspberrypi-sys-mods wants to install systemd

Wurzelmann, 2018-01-15:
The new version of raspberrypi-sys-mods is trying to install systemd and I'm not sure why this is necessary.

Depends: init-system-helpers (>= 1.18~), libcap2-bin, systemd (>= 230)

Is there any way around this as I'm not keen on installing systemd at all.

spl237, 2018-01-15:
systemd is the init system used by both Debian Jessie (which is now outdated) and Debian Stretch (the current release). sys-mods therefore assumes the presence of systemd. If you are still using Debian Wheezy or earlier, this release of sys-mods really isn't going to work properly for you anyway - if you are on a version of Raspbian released in the last two years, it will have systemd installed; if you aren't, there's not much point updating to the latest sys-mods.

closed 2018-01-15 by spl237

XECDesign, 2018-01-15:
Although Debian doesn't force systemd, we are not going to maintain support for multiple init systems in our packages. The workaround would be to take the parts of sys-mods you need and apply them in a way that works with the init system of your choice.

Wurzelmann, 2018-01-15:
Our machine is running Debian Stretch, but we're not using systemd and we did not realize this was going to be a problem with raspberrypi-sys-mods...

Could you be so kind as to point me in the right direction of the parts we're going to build our workaround with?

Thank you in advance.

XECDesign, 2018-01-15:
All of it, but the part that would need to be ported over would be the systemd services in the `debian` directory.