RPi-Distro/raspberrypi-sys-mods #23: sourcing of gettext.sh causes error message at login

mapausch, 2018-03-22:
with installed raspberrypi-sys-mods the error message "-bash: gettext.sh: Datei oder Verzeichnis nicht gefunden" comes up after successful login.

("Datei oder Verzeichnis nicht gefunden" translated to english: File or directory not found )

tracked the message down to /etc/profile.d/wifi-country.sh where in line 4
the file gettext.sh is sourced:

```
(
        export TEXTDOMAIN=wifi-country

        . gettext.sh
```

spl237, 2018-03-23:
gettext.sh is required for multi-language support. I can repeat the error message you show by deleting the file /usr/bin/gettext.sh, so it has presumably either been deleted or not been installed on your system.

"sudo apt-get install --reinstall gettext" should fix it.

spl237, 2018-03-23:
Hang on - that seems not to reinstall the .sh file - I'll try to find what installs that.

spl237, 2018-03-23:
"sudo apt-get install --reinstall gettext-base" should fix it.

closed 2018-03-23 by spl237