RPi-Distro/raspberrypi-sys-mods #25: Udev rules on arm64

satmandu, 2018-06-28:
Would it make sense / be possible to change etc.armhf to just be etc so that it is installed on both armhf & arm64? Or alternatively just create a separate etc.arm64 subtree alongside the etc.armhf subtree for the files that work just fine on arm64?

Foe instance the etc.armhf/udev/rules.d/99-com.rules  file definitely works on arm64, and is required to setup the serial device node so that the Bluetooth uart device scripts work in arm64 installed on the 3b+.

XECDesign, 2018-06-28:
No, since that stuff would be added to x86 builds, which would be undesirable.

I wonder if a symlink would do the job.

satmandu, 2018-06-28:
As in symlink debian/raspberrypi-sys-mods.install.armhf to debian/raspberrypi-sys-mods.install.arm64 ?

XECDesign, 2019-04-29:
> As in symlink debian/raspberrypi-sys-mods.install.armhf to debian/raspberrypi-sys-mods.install.arm64 ?

Going over old issues and I have no idea what I meant there, but that's probably it.

commit d9c1864b (2019-07-22): Add raspberrypi-sys-mods.install.arm64

Fixes #25
closed 2019-07-22 by XECDesign