RPi-Distro/raspberrypi-sys-mods #26: /usr/lib/raspberrypi-sys-mods/wifi-country incompatible with recent kernels

larsks, 2018-12-17:
I'm running kernel 4.19.8-v7+ on a Pi 3B+ (from `BRANCH=next rpi-update`).  When running `systemctl status`, I noticed that the system was in a "degraded" state.  Running `systemctl list-units --failed` showed that the failed unit was the `wifi-country.service` unit.

That service runs `/usr/lib/raspberrypi-sys-mods/wifi-country`, which performs the following logic to get the WiFi iomem address:

    address="$(/bin/grep -m 1 /mmc /proc/iomem | /usr/bin/cut -f1 -d-)"

The contents of `/proc/iomem` look something like this:

```
root@raspberrypi:~# cat /proc/iomem 
00000000-3b3fffff : System RAM
  00008000-00bfffff : Kernel code
  00d00000-00e69ecf : Kernel data
3f006000-3f006fff : dwc_otg
3f007000-3f007eff : dma@7e007000
3f00b840-3f00b87b : mailbox@7e00b840
3f00b880-3f00b8bf : mailbox@7e00b880
3f100000-3f100027 : watchdog@7e100000
3f101000-3f102fff : cprman@7e101000
3f104000-3f10400f : rng@7e104000
3f200000-3f2000b3 : gpio@7e200000
3f201000-3f201fff : serial@7e201000
  3f201000-3f201fff : serial@7e201000
3f202000-3f2020ff : mmc@7e202000
3f212000-3f212007 : thermal@7e212000
3f215000-3f215007 : aux@7e215000
3f300000-3f3000ff : mmc@7e300000
3f980000-3f98ffff : dwc_otg
```

That grep for `/mmc` will never match, and running the `wifi-country` script results in:

    Could not determine WiFi iomem address

pelwell, 2018-12-19:
That's an interesting find. Investigating has uncovered the fact that when it worked it was actually matching the wrong mmc interface, meaning that the rfkill wasn't actually blocking anything. See #27 for a fix for both issues.

larsks, 2018-12-19:
Looks good. I guess I'll close this one :).

closed 2018-12-19 by larsks

pelwell, 2018-12-19:
Does that mean you've tried it on 4.19? Some confirmation that it works as expected would be welcome.