RPi-Distro/raspberrypi-sys-mods #32: Buster gpio udev rules

bcroston, 2019-02-17:
In the Debian-maintained Buster package for RPi.GPIO, the package maintainer has included the udev rules in the 'rpi.gpio' debian package.  It uses the 'dialout' group rather than the 'gpio' group.  Could there be a conflict with 'raspberrypi-sys-mods/etc.armhf/udev/rules.d/99-com.rules'?  It suspect it will cause a lot of confusion with end users following existing guides found around the web.

XECDesign, 2019-02-19:
It's worth testing, but the sys-mods version should run last and take priority over the version shipped in rpi.gpio. So, the old behaviour should be preserved within our Raspbian images.

XECDesign, 2019-07-22:
Okay to close?

bcroston, 2019-07-22:
It seems to work fine now we have done a release

closed 2019-07-22 by bcroston

XECDesign, 2019-07-22:
Thanks, Ben.