RPi-Distro/raspberrypi-sys-mods #33: 010_pi-nopasswd change

RPIWS, 2019-05-30:
Would be great if you instead of pinpointing a user in 010_pi-nopasswd you instead
pinpoint a group for example like this: 

010_rpi-nopasswd:

%magicwand ALL=(ALL) NOPASSWD: ALL

as user pi anyway in my case its the first i remove from raspbian and create my own, so 
hopefully you would consider altering this to a group policy instead of user policy.
I could make my own deb package with the correct settings i want but feel its better if upstream version of raspberrypi-sys-mods would have this feature instead :)

//Martin aka ztealmax - rpiws.org

XECDesign, 2019-07-22:
I don't think I can justify messing with sudoers for this. The standard reasons not to change things seem to outweigh the benefits here.

RPIWS, 2019-07-22:
So hmm you think its ok to keep user pi is allowed to be sudoer even if the
user account isnt on the system anymore (a user everyone knows the name of
and targets), i really dont get it why it should be a problem to have a
group instead of a specific username

XECDesign, 2019-07-22:
No, I think when a user removes `pi`, they shouldn't adjust the nopasswd file accordingly (or remove it) as a part of the same process.

There are many configurations out there that I can't think of that making a change like this will upset, since it's a sensitive area.
There are people who would've modified the file and wouldn't want us to mess with it. There are people who will already have the group added and then suddenly that group gets `NOPASSWD`. Many people just won't like sudoers being modified by the system without their explicit permission.

It's not a trivial change and would require maintainer scripts to add the group if it doesn't exist. Check the current state of things and then act accordingly. I've been burned by treating this too lightly before.

spl237, 2019-07-22:
If the pi user isn't on the system any more, it doesn't matter if it has sudo access - because you can't log in as it, as it doesn't exist. You need sudo access to create a new user, or to reinstate the pi user, so whether or not you remove sudo access from a non-existent user account makes no difference to the security of the system.

My colleague above is correct - the risks attached to this change outweigh the (largely non-existent) advantages.

closed 2019-07-22 by XECDesign

RPIWS, 2019-07-22:
WEll problem is everytime system updates it replaces it back to user pi :/

RPIWS, 2019-07-22:
Nevermind i just block raspbian-sys-mods and install my own repo, thanx...

Den mån 22 juli 2019 kl 15:47 skrev Martin Svensson <
martin.svensson@ztealmax.se>:

RPIWS, 2019-07-22:
Just as a side note, Arch Linux uses same policy as i describe and it works
like a charm so i really dont get your point of "messing with sudo" you are
just narrowing it down for the user experience

Den mån 22 juli 2019 kl 15:48 skrev Martin Svensson <
martin.svensson@ztealmax.se>:

RPIWS, 2019-07-22:
Ohh btw describe the risks to a fellow engineer so i understand your point
of view more thank you

Den mån 22 juli 2019 17:49Martin Svensson <martin.svensson@ztealmax.se>
skrev:

XECDesign, 2019-07-22:
> WEll problem is everytime system updates it replaces it back to user pi :/

It's a conffile, so no, dpkg won't touch it in this case.

>  i really dont get your point of "messing with sudo" you are
just narrowing it down for the user experience

I highly doubt Arch has any group with NOPASSWD permissions, and it's certainly not done automatically under the user's nose. The standard setup is to have a `wheel` or `sudo` group without `NOPASSWD`, which is also the case in Raspbian (in addition to user `pi` having `NOPASSWD`).

There's nothing preventing anybody from using the system the way the user sees fit in this regard. This request isn't about expanding the user experience, but automatically transferring everybody to your preferred setup.

If you would like to discuss this further, the forum would be the more appropriate place.