RPi-Distro/raspberrypi-sys-mods #41: Where is the 20210125 update ??

FredericGuilbault, 2021-02-04:
It is signed by   — Serge Schneider Mon, 25 Jan 2021 16:03:24 +0000

But I don't find it anywhere in :

https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/master/debian/changelog

Im confused, is this voluntary ?

FredericGuilbault, 2021-02-04:
The .dsc control file stipulate that source repository of the package is your  github repository. 

```
Vcs-Browser: https://github.com/RPi-Distro/raspberrypi-sys-mods
Vcs-Git: git://git@github.com:RPi-Distro/raspberrypi-sys-mods.git
```

http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-sys-mods/raspberrypi-sys-mods_20210125.dsc

XECDesign, 2021-02-04:
I don't generally push things to git immediately, but it should be there now.

closed 2021-02-04 by XECDesign

FredericGuilbault, 2021-02-05:
@XECDesign  Any plans to add repositories in pi-get instead of post-inst scripts, This is for the least.  ...not the most iconic way to add a repository to a distribution.

OmgImAlexis, 2021-02-05:
> I don't generally push things to git immediately, but it should be there now.

You should really look into changing this practise. The source should be pushed and viewable before updates go out.

robertoschwald, 2021-02-10:
... and do not enable the repo by default. 
This smells.