RPi-Distro/raspberrypi-sys-mods #43: Remove Microsoft vscode from raspberry pi os lite

sabotagebeats, 2021-02-04:
Hello
I have headless pi and do not want vscode or any Microsoft products. 
How can I remove it?
Thanks

sabotagebeats, 2021-02-04:
Do I need to wipe the pi and install a different OS?

sabotagebeats, 2021-02-04:
https://github.com/RPi-Distro/raspberrypi-sys-mods/commit/655cad5aee6457b94fc2336b1ff3c1104ccb4351#r46761662

sabotagebeats, 2021-02-04:
https://github.com/RPi-Distro/raspberrypi-sys-mods/commit/655cad5aee6457b94fc2336b1ff3c1104ccb4351#r46761672

sabotagebeats, 2021-02-04:
I have done the following. What else do I need to do? 

`sudo rm /etc/apt/sources.list.d/vscode.list
sudo rm /etc/apt/trusted.gpg.d/microsoft.gpg`

pelwell, 2021-02-04:
Just delete the additional entries in the sources - now that the version number has passed that particular value it won't be added again.

pelwell, 2021-02-04:
And maybe give people a chance to answer?

ricardoandren, 2021-02-04:
And maybe give people the option to install it instead of forcing it onto people?

pelwell, 2021-02-04:
Have you read the other comments here? Do you see the constructive dialogue that some users manage? The ones actually doing something useful to achieve their goal? Be more like them, and stop knocking on an open door.

ricardoandren, 2021-02-04:
So saying "And maybe give people the option to install it instead of forcing it onto people" isnt constructive? I dont see whats not constructive about it.

1:
how can you expect us to be constructive when you stealthly add a repo thats unwanted for most of us? 
2:
How can we expect you to be serious when even Eben Upton doesnt think its serious
[image]
3:
Its not like the dev team didnt know there would be a backlash against putting something from Microsoft into Linux.

pelwell, 2021-02-04:
Unconstructive comments will be deleted.

ricardoandren, 2021-02-04:
> Unconstructive comments will be deleted.

More sabotage attempts.

XECDesign, 2021-02-04:
> 
> 
> I have done the following. What else do I need to do?
> 
> `sudo rm /etc/apt/sources.list.d/vscode.list && sudo rm /etc/apt/trusted.gpg.d/microsoft.gpg`

Throw a `sudo apt update` and you're set.

closed 2021-02-04 by XECDesign

FredericGuilbault, 2021-02-05:
>  Throw a sudo apt update in there as well and you're set.

@XECDesign  No, Unless im wrong,  on the next update of this package. The repository will be re-added. 

If @sabotagebeats only remove the file.  The only thing this user is "set" ATM is to allow the re-installation of the microsoft repo as soon as postinst will be run agan. 
 
 
  
https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/655cad5aee6457b94fc2336b1ff3c1104ccb4351/debian/raspberrypi-sys-mods.postinst#L68

This is line only check if file exist, If not, the repository is re-added everytime the user run  `dpkg-reconfigure -a`, or  a new version of raspberrypi-sys-mods is released.
 

___

On the current state of the code. The file must remain.  After deleting the unwanted repo. The user can use  

`apt-mark hold raspberrypi-sys-mods ` if the user doesn't want this package to mess again with it's repo list. 
Sadly, this package is an omnibus bill, it does not _"do one thing and do it well"_.  the user would also cut itself from receiving others changes he might want.

The hacky way to do would be to do `touch /etc/apt/sources.list.d/vscode.list` not the cleanest solution but what can a user do at this point... ? 

For more security and confidence, I would also recommend doing `chattr +i /etc/apt/sources.list.d/vscode.list` to make sure that any bad regex in the future decide to update the content of the file.

____
I begin to understand why people feel that this update is forced on them and sneaky. 

I can't believe that no one in your team have flagged that would be a sensitive update and extra care would have to be taking.... knowing how much Linux people love to hate M$ since 1991 .

FredericGuilbault, 2021-02-05:
Also the title is : 

> Remove Microsoft vscode from raspberry pi os lite

and I don't see it as solved.      Is it a won't fix ?

FredericGuilbault, 2021-02-05:
> now that the version number has passed that particular value it won't be added again.

@pelwell I think this is false 
  
https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/655cad5aee6457b94fc2336b1ff3c1104ccb4351/debian/raspberrypi-sys-mods.postinst#L122

The function will be executed on every update in the future, 

and as stated earlier, if the user delete the file. 

https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/655cad5aee6457b94fc2336b1ff3c1104ccb4351/debian/raspberrypi-sys-mods.postinst#L68

Will add it agan.

pelwell, 2021-02-05:
The version comparison line is checking if the existing version is older than "20210125", and only running add_ms_repo if it is. Once you have gone past that version of won't try again.

FredericGuilbault, 2021-02-05:
I tested it and you are right.  I have read lt-nl backward I guess, My bad.

pelwell, 2021-02-05:
We are actively reviewing the distribution mechanism, having received a number of helpful suggestions, but there is at least a simple way to opt out.

RUSshy, 2021-02-05:
> but there is at least a simple way to opt out.

This should be the opposite, "a better way to opt-in"

But anyways, reputation for this project is forever lost, i used to recommend raspberry to people, that won't be the case anymore

c33s, 2021-02-10:
what about the fact that you are potentially violating the GDPR law in europe? as i already commented on the PR https://github.com/RPi-Distro/raspberrypi-sys-mods/pull/51#discussion_r571708157 there is a GDPR problem here.

to sum this up: you are risking law problems for you and your users in europe and you are losing your communities trust just to spare a small subset of your users this commands:

```
wget -qO- https://packages.microsoft.com/keys/microsoft.asc | gpg --dearmor > packages.microsoft.gpg
sudo install -o root -g root -m 644 packages.microsoft.gpg /etc/apt/trusted.gpg.d/
sudo sh -c 'echo "deb [arch=amd64,arm64,armhf signed-by=/etc/apt/trusted.gpg.d/packages.microsoft.gpg] https://packages.microsoft.com/repos/code stable main" > /etc/apt/sources.list.d/vscode.list'
```

> Given that the repo doesn't do anything unless she chooses to install vs code, I'd advise against risking somebody else's life for your ideology.

@XECDesign https://github.com/RPi-Distro/raspberrypi-sys-mods/pull/52#issuecomment-776664907

it is nothing about ideology it's about law. by installing the repo without an active consens of the user you are simply acting against this law as each `apt-get update` pings the m$ server and exposes their ip to them.

> ... but there is at least a simple way to opt out.

@pelwell https://github.com/RPi-Distro/raspberrypi-sys-mods/issues/43#issuecomment-774072654

it has to be opt-in and not opt-out.

> Since our aim is to remove hurdles and make things simple for beginners who may not know what a repo is, requiring input like that is just confusing.

@XECDesign https://github.com/RPi-Distro/raspberrypi-sys-mods/pull/51#issuecomment-776611112

as i wrote above you see the command it needs to install vscode. we are talking about software development. if the persons which will use vscode are confused by going to the shell and enter the required commands to install or answer a simple y/n dialog, maybe they should not develop at all? at least admit that it's about money and a deal you have with microsoft.

if you want to help them simply add a shortcut to the deskop which starts a script which is doing running the command for them but installing a third party for all of your users is simply wrong.

you don't really remove hurdles in fact your are build up a very big hurdle for all people in europe who want to use this OS legally.

for a small subset of persons which want to use vscode you make it easier and make it more difficult for everyone in else europe. yes most of the people simply don't care if it is legal or not until they get sued. a associate of a customer of mine had to pay 5000eur just because of a GDPR violation (one person company).

with opt-out instad of opt-in you will require a privacypolicy which the user has to accept before they can download the image. in that policy you are required to inform the user that their ip is sent to mircosoft.

i really don't understand this reaction of you (the company who creates the raspberry OS): you are really losing your communities trust, your force close discussions, ... 
don't you see the $hitstorm you created with this? simply go one step back and remove all third party repos.

XECDesign, 2021-02-10:
The decision to add the repo came from people who are more familiar with GDPR than I am, so I'll trust their judgement.

RUSshy, 2021-02-10:
> The decision to add the repo came from people who are more familiar with GDPR than I am, so I'll trust their judgement.

Microsoft's salesman?

They are wrong, and you are trusting the wrong people

> Like the DPA, the GDPR applies to ‘personal data’. However, the GDPR’s definition is more detailed and makes it clear that information such as an online identifier – e.g. an IP address – can now be considered as personal data.

So in short, remove the repo or explicitly mention IP will be shared with Microsoft in a user-agreement form

XECDesign, 2021-02-10:
If you believe we're mishandling personal data in any way, I wouldn't discourage anybody from protecting their rights:

https://edpb.europa.eu/about-edpb/board/members_en

https://ico.org.uk/make-a-complaint/

starbasessd, 2021-02-10:
I don't believe YOU are, but you are enabling m$ collecting data on users of other devices and OSs without asking those users first (especially non-GUI users as VSCode is a GUI app).

BitBistro-code, 2021-02-10:
I've reviewed the data, ala tcpdump, sent back and forth and there is no cookie exchange. There is a user agent which is complete standard. There is also an etag which is very different than a cookie.
Of course the service owner will have logs of each hit. When you visit a website they have that log whether or not you click the accept cookies button.
So I guess I don't know where the line is drawn as far as data collection and GDPR. As far as I can tell they have service logs only.

starbasessd, 2021-02-10:
So Microsoft, who has a massive database, can't then associate the log of
the hit to packages.microsoft.com, find out which users of m$ stuff are now
using RPi stuff? Or is that naive?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BitBistro-code, 2021-02-10:
They can as anyone could. Without the cookie they would not be able to target the individual assets as accurately and in some cases not at all. 
So we are talking about anonymous aggregated data. If that is a GDPR violation and they are doing it they should stop. I think that falls on Microsoft if that is the case. I'm no lawyer though and I have not studied the law in detail other than what my employer requires me for my job.
I agree with @XECDesign that a complaint should be filled if you believe there is a mishandling of person data.
I just wanted to let you know what data is actually being exchanged. If you wish I could probably teach you how to run tcpdump and read some simple http packets.

c33s, 2021-02-11:
it is not about cookies it is about personal data, where the ip counts towards personal data:

> What is personal data?
>....
>an Internet Protocol (IP) address;
>...

https://ec.europa.eu/info/law/law-topic/data-protection/reform/what-personal-data_en

if some university mirror gets my ip where the chance is high that they don't aggregate the personal data, well, simply does not matter but microsoft is doing telemetry and is quite aggressive with their data collection. so no i don't want them to get my ip. so yes i still think it is a GDPR violation.

if i connect to a webpage, i know that i share my ip, if i install debian i know that the debian mirrors get my ip but no i don't assume that if i install **linux** my ip gets to microsoft.

starbasessd, 2021-02-11:
And I don't see Debian, or RedHat, or Ubuntu or other linux Distros asking m$ to put their package update sites into the base m$ installs for any reason.  Maybe, RPiFoundation is going to get reciprocity, to know where and how many m$ installs there are? Not necessarily cookied, but just the hits from IP Addresses would tell them much... Where m$ can and does aggregate and correlate all their data, users, installs, locations, and now can get other systems. 
If you own a MAC, but don't install any m$ software, does your MAC check with m$ sites?

BitBistro-code, 2021-02-11:
I apologize in advance if this response exacerbates this discussion. I feel that this is now a closed issue and not the correct medium to discuss this further. This will be my last update to this thread. 

An IP address alone without other identifiers (such as cookies) is considered to be psuedo-anonymized, and does not qualify as personal identifying information under GDPR. This is classified as essential data to be stored for the health and safety of the servers that are being accessed. Assuming that they are used for the purpose of security and maintenance.

See also recital 49 of GDPR:
https://eur-lex.europa.eu/legal-content/EN/TXT/?uri=CELEX:32016R0679

You can't expect a website to purge you server logs just because you clicked no to the server cookies. Honestly, I don't even think that APT would know how to store and pass the cookies back and forth. I do not believe there is a violation here.

I for one am not actually in favor of adding this repo by default, but for different reasons. Regardless, I do not feel this is a rights or privacy violation and will just remove the repository on the servers that I do not want it on. There are more important causes for me to allocate my time to. I also do not believe a positive outcome will be had by discussing it further here. I think that @XECDesign has heard the communities response and they are discussing internally what to do next.

If you just want to argue. I love to argue. Send me an email and we discuss. It is pretty boring sometimes being stuck inside. Lots of snow here in New England. Reddit is a good place to argue too.

XECDesign, 2021-02-11:
I think you hit the nail on the head, but the argument from others is that they can combine that IP with other information they've collected. You may have accepted some EULA that lets them track you while signing up for some other MS service like Azure or Github, or maybe while installing something you don't even remember. Now they can run those IPs by the HTTP access logs and determine who's who.  This way they'd know that you're probably running Raspberry Pi OS and when you're active and that may contribute to an overall profile they may or may or may not have. Maybe someone might say, "okay I don't care if they track me on github, but I don't want them also knowing what else I'm doing". At least, that's my best effort to steel-man the argument.

Then there's also the hypothetical possibility that they're logging and cross-referencing everything, without checking who has opted in or out. Or maybe they're sharing those logs with other partners which have data they can cross-reference. If something like that's happening, they could be a violating some laws. I am sure that they have lawyers which they have to run everything by.

And yes, we've been keeping an eye on all the feedback, forum, reddit posts, youtube videos and so on, but there are things to sort out internally become we can make a final decision.

nanotuxi, 2021-02-12:
This discussion is rediculous. Since the early days of linux it is my favorite os. A lot of discussions between linux lovers and ms enthusiasts have been read by me. 
Fact is that microsoft is swallowing open source projects for a while now. This platform here (github) is the best known example for this. 
I don't know how many people spent years of their lifetime to develop excellent software. More than 90% of the internet server infrastructure id based on linux. And all the old guys like me know that MS hated linux in the early days because it is just more flexible, robust and secure. 
If now a company like raspbery pi comes along with such a b..shit patch it is like kicking all those people into there a.. 
So my final decision is... good by raspberry pi. There are tons of other hardware platforms we can use and we will use.

FredericGuilbault, 2021-03-06:
@pelwell @XECDesign  

 Any decision taken at raspberry foundation ?

DanieleQ97, 2022-09-20:
@BitBistro-code 

> This is classified as essential data to be stored for the health and safety of the servers that are being accessed._

> You can't expect a website to purge you server logs just because you clicked no to the server cookies.

It's true that Microsoft has every right to keep a record of just my IP address making a connection to some their server.
The point here is **"Why is my Raspberry making a connection to some Microsoft server every time I update it even though I've no intenction of installing any Microsoft product on it?**.

I can't expect a website to purge their sever logs if I opt out of their cookie policy, but I can expect them to NOT put in their homepage a 1-pixel transparent png hosted on the server of some completely unrelated company so that the completely urelated company can keep a record of my IP visiting that homepage.