RPi-Distro/raspberrypi-sys-mods #48: Fix WPA2 Enterprise Wifi Module

Picajoluna, 2021-02-05:
Hello,

I'm a college student who owns a Raspberry Pi Zero W. I've struggled to get my RPi connected to my school's WPA2 Enterprise network, until I very recently came across this [fix on a forum post](https://www.raspberrypi.org/forums/viewtopic.php?p=1601893&sid=9154e8b5618a5ddb5054191a15ea3af4#p1601893), which involves a very simple swap of `nl80211,wext` for `wext,nl80211` in the `/lib/dhcpcd/dhcpcd-hooks/10-wpa_supplicant` system file.

It took me forever to find this fix, until then I was using a script that I found elsewhere to downgrade the wpa_supplicant package to an older version, which is not ideal.  I would love to issue a pull request to suggest the change and prevent other folks this needless headache, but I can't find the GitHub repository which contains that file / the repository in which I can issue the PR. Does anyone know where I could start that conversation, if this is the wrong place for it?
Thanks.

BitBistro-code, 2021-02-10:
I think this would be more suited for the raspberrypi-net-mods package.

XECDesign, 2021-02-10:
I seem to recall that having it swapped with wext first also results in something else not working, but unfortunately I don't recall what exactly. I believe this will just swap one set of issues for another, so I'd like to avoid a change in behaviour like that.

Picajoluna, 2021-02-13:
Hm, sorry to hear that XECDesign. I'd be happy to post over in raspberrypi-net-mods – I just wish that this issue had been resolved for me when I started trying to connect to university wifi. Instead, I had to go hunting on forum posts and only recently found this fix for the raspberry pi zero w... Feelsbad for college students I guess^^' I'd love to have this resolved for folks going ahead.