RPi-Distro/raspberrypi-sys-mods #49: purge /etc/apt/sources.list.d/vscode.list how?

alexxroche, 2021-02-07:
I don't use vscode. I will NEVER use vscode. I run my raspberry pi headless as a DNS filter for an education environment. (Where we use libreoffice.) I only access RPi via ssh over openVPN.

How do I instruct apt to remove vscode.list and other trojans, such as /etc/apt/trusted.gpg.d/microsoft.gpg (that seem to have appeared on my devices without my request)? [Seem ironic to have a corporation's key inserted into "trusted" when this just confirms that they can't be - double irony for me reporting here on github.]

I would also like to know the best way to prevent ANY Microsoft files being inserted into /etc on my linux operating system. (Feels worse than a root-kit.) While I sensibly and calmly investigate the potential impact of migrating to either archlinuxarm or FreeBSD.

FredericGuilbault, 2021-02-07:
it's going to be hard to remove/purge or hold that package or this specific update. Cuz it also contains others important changes (mainly the change or raspberry repository URL).   You will have to act at file level.

vaporwave9, 2021-02-09:
nuke it
create an empty file with same name
lock it forever from being re-created

```
sudo rm -vf /etc/apt/trusted.gpg.d/microsoft.gpg
sudo touch /etc/apt/trusted.gpg.d/microsoft.gpg
sudo chattr +i /etc/apt/trusted.gpg.d/microsoft.gpg
```
When trust has been restored you can undo the lock. But why would we ever want a Microsoft key on our headless servers?
After these steps they cannot push the untrusted code anymore.

alexxroche, 2021-02-09:
> why would we ever want a Microsoft key on our headless servers?
This was my thought. As a linux operating system, adding Microsoft made no sense to me. (Am I the only one that is happily living in a pure unix world?) I have Linux for my desktop, BSD for my servers, (and after this fiasco we have), archlinuxarm for our last raspeberrypis.

If others want to add vscode then they could add the repo and the key themselves. (This is the debian way.) Forcing a repo and key onto ALL raspberrypi-sys-mods because _some_ people **might** like vscode seems selfish. (Not to mention pointless bloat; one more thing to monitor; a total destruction of trust.) To give you an analogy: We support people's right to be gay; we don't force everyone to be gay. Do you see the difference? Making things available to people is very different from thrusting it into them without consent.

I understand vaporwave9's frustration, but I don't want to "nuke it", (though I have had to.) I want those that thrust Microsoft into my copy of this non-Microsoft operating system, without my consent to realise that "they have made a mistake." It would be clear that they realised their error if they cleaned up after themselves. They spilt this mess on our machines, so they should be the ones to clean it up. (And if they "can't" that's a clear indication that they should have done so in the first place.) This would be the first step in rebuilding trust with their community. By way of an apology they could maybe add some oversight? (As a rule, If you're not sure, then just ask. If you **are** sure that making the addition of vscode.list mandatory for all, was a good idea, then your confidence meter is broken, (or your friends are lying to you, (or vaporwave9 and I are in a very limited bubble.))

I'm sorry that I can't fully express my anger and disappointment, but that would require some uncivilised words, which I'm not prepared to post in a public forum.

vaporwave9, 2021-02-09:
Well since there is a clear distinction between "Lite" and "Desktop", from a very logical and emotionless standpoint, why bundle a visual editor in the headless distros anyway?
https://www.raspberrypi.org/downloads/raspberry-pi-os/

But otherwise I agree with you fully.

> but I don't want to "nuke it"
> 

My bad, I was just trying to give a quick though not perfect solution.

> or vaporwave9 and I are in a very limited bubble
> 

Well, it is primarily a SBC for young people and aspiring programmers right?
I mean I can totally understand if I sound jaded by invading their ecosystem.

Raspberry is just loved across every age and culture. Sure the "old" people with their servers want to have a say too :) 
I don't even dislike MS but as @alexxroche said:

> living in a pure unix world

Is an ideal we should also tech young people. Isn't that what Raspberry stands for as well?

/rant over

I let the repo owners handle this now, but I hope me being too chatty here won't get it locked up as well.
Discussions about ideologies don't belong in issues, I agree. But this is an issue of trust, if not just a technical one.
An issue of Free Software and the *NIX thought.

FredericGuilbault, 2021-02-09:
> "nuke it"  

Technically, as the current state of the code in the package. You don't have to `chattr`   Just removing the file is enough (if it have already been added)    people use `chattr` by lack of trust of their next commits mainly.

closed 2021-02-10 by XECDesign

alexxroche, 2021-02-11:
Why was this closed? Neither
> How do I instruct apt to remove vscode.list and other trojans, ...

nor
> I would also like to know the best way to prevent ANY Microsoft files being inserted into /etc ...

have been answered. Should I create a separate ticket for each? (I felt they were related enough that they could both be dealt with at the same time - but I quite accept that I could be mistaken.)
If it wasn't clear: 
1. I do not want ANY files being silently added to /etc/apt [least of all Microsoft specific] on my headless raspberry pi. (It doesn't matter how "innocuous" you think it is, it is an abuse of trust.)
2. If raspberrypi-sys-mods polluted my machine without asking, would it seems fair for you to clean up your mess?
3. Could we have some assurance that this will NEVER happen again? (Or is this, "Don't care, won't fix!" ?)

MichaIng, 2021-02-12:
> How do I instruct apt to remove vscode.list and other trojans, ...

> I would also like to know the best way to prevent ANY Microsoft files being inserted into /etc ...

There is no way to instruct APT to do so or assure otherwise that DEB packages you install do so via their postinst script. Since installing a DEB package requires root permissions and there are not really limits what postinst scripts can do, you need to first review or trust packages before you install them.

In case of the Debian repository (and practically Raspbian as well, since the sources are the same (?)), their own policies, review and testing mechanisms are a good reason to trust them. One might say, trust is not needed, as the package sources can be reviewed, if one sticks with the "main" open-source component, but strictly there is no guarantee that the served packages were really build from those sources, like it was with the `raspberrypi-sys-mods` package not matching this GitHub repository the first day the MS repo was added. So it's Debian and their own policies you practically still need to trust in. But the Raspberry Pi repository obviously doesn't have such policies, beginning with the obvious fact that it serves closed-source and/or non-free software, including the RPi firmware itself. Just keep in mind that, if you use Windows, macOS/iOS or any of most vendors Android phones, you are using MUCH larger closed-source non-free software blobs + firmware + hardware already, so no need to treat RPi unfair in this regards.

So if you want assurance, you should use an open-source SBC that runs with the mainline kernel (served by the Debian repository as well, or Armbian as an alternative) and that at best is well supported by the Mesa open-source GPU drivers, if required. But what I can assure you is that you won't have the rich hardware feature support and a much smaller (SBC/SoC-specific) community (yet), so it comes with it's undoubted downsides IMO, depending on the way you actually use the board.

I think what you are actually waiting for is a statement/reaction from the RPi guys that re-builds your trust in them, I do as well. I read elsewhere that they do read all reactions and discussing it internally. Let's see what they come up with. I think every result that does not include an interactive admin permission will keep it a large reputation/trust damage at this point. I read no single statement of an actual user saying something like "Thanks for adding this repo, I'm now enjoying MS VS Code on my RPi" 😄.

alexxroche, 2021-02-13:
> raspberrypi-sys-mods (20210208) buster; urgency=medium
> 
>   * Stop-gap measure to address one of the main concerns about 3rd party repos
>     - Prevent VS Code repo from potentially overriding system packages
>     - Only allow installation of known packages (code-*)
> 
>  -- Serge Schneider <serge@raspberrypi.com>  Mon, 08 Feb 2021 12:37:18 +0000

Rather than putting a bullet-proof napkin on the unexploded bomb that you dropped into our lives, maybe... just remove the bomb? (And apologise?) Just a thought.