RPi-Distro/raspberrypi-sys-mods #50: Separete the management of  /etc/apt/sources.list.d/vscode.list in a separate package.

FredericGuilbault, 2021-02-07:
Still I don't know where you are about this problem, ill keep suggest new possibility. 

This one would be to separate the management of the file  /etc/apt/sources.list.d/vscode.list and the related key management in a separate package that do one thing (adding M$ repo) and do it well.

jpmcneely, 2021-02-09:
This is the correct solution.  Make it a package.  There is no need to make this part of the existing sys-mods.

XECDesign, 2021-02-10:
That's not one of the options we're considering.

closed 2021-02-10 by XECDesign

FredericGuilbault, 2021-02-10:
Thx, at least that make it clear. So we know now.  Silence is often the worst than unpopular answers.

RubenNL, 2021-02-10:
> That's not one of the options we're considering.

Why not? What makes a postinstall script superior to a package?