RPi-Distro/raspberrypi-sys-mods #53: Dedicated LTS Raspberry Pi OS image for Compute Modules.

mariusmotea, 2021-02-22:
Hi,

According to the [official blog](https://www.raspberrypi.org/blog/raspberry-pi-compute-module-4/) there is following statement:

> Over half of the seven million Raspberry Pi units we sell each year go into industrial and commercial applications, from digital signage to thin clients to process automation.

Most industrial applications use the os lite version with no desktop interface at all and are behind a firewall that monitor the connections. When an OS pushes a package update and without user permission, install some trusted keys and create new hits on the firewall that require an ITSEC investigation, well, this has nothing to do with the industrial domain. Basically **CM4 module remains out of scope** in my opinion and the only way to recover the lost trust is to have an LTS version with the guarantee that no bloatware will be added. Connections management is a must for today security standards, security devices trigger an alert or totally block a device if unexpected traffic is detected as the device will be considered compromised. 
Things are even worse because I had not seen any acknowledgement that this was a bad decision, but instead comments to previous issues are blocked and some of them are even deleted. 

PS. I know I can delete these two files, but imagine you need to do this on dozens devices.

pelwell, 2021-02-22:
The trouble with including a grievance inside what might otherwise be a reasonable request is that the recipient is likely to only see the grievance. Try again.

closed 2021-02-22 by pelwell

mariusmotea, 2021-02-22:
Thanks fot your reply. I will consider other distros for my projects.

starbasessd, 2021-02-22:
You could consider using pi-gen (https://github.com/RPi-Distro/pi-gen is what THEY use to generate the image) to build your own image(s), thus having better control over what is or is not included, and how it's included. The scripting is easy to follow, modify, and generate any or all3 primary images. Easy to add your standard packages, pre-remove what you always remove, modify, change sources, etc. I immediately went into mine and removed this, since I did not see anything happening here that I needed. I also added my own vs.list and m$ 'trusted' file, both blank, and made them immutable. It's a moving target, I know, but like a virus, I fight things when I can, how I can. My 'lite' versions have absolutely no need for that feature, and now that they've revealed a security hole, I will apply due diligence (as well as switch distros) as I am able.