RPi-Distro/raspberrypi-sys-mods #61: Post-installation script subprocess returned error exit status 126

JanWennrich, 2022-01-13:
I just executed an `sudo apt update` followed by `sudo apt dist-upgrade` on my `Raspberry Pi 3 Model B Plus Rev 1.3` running `Raspbian GNU/Linux 11 (bullseye)`

The upgrade failed with the following error message:

```
Setting up raspberrypi-sys-mods (20220110+1) ...
Fixing previous dhcpcd wait.conf drop-in to prevent double logging...
/var/lib/dpkg/info/raspberrypi-sys-mods.postinst: 27: sed: Exec format error
dpkg: error processing package raspberrypi-sys-mods (--configure):
 installed raspberrypi-sys-mods package post-installation script subprocess returned error exit status 126
Errors were encountered while processing:
 raspberrypi-sys-mods
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Checking `/var/lib/dpkg/info/raspberrypi-sys-mods.postinst:27` I found that the script tries to replace something in the `/etc/systemd/system/dhcpcd.service.d/wait.conf` which contains the following content on my device:

```
[Service]
ExecStart=/usr/sbin/dhcpcd -w
```

Is my `wait.conf` messed up or is something wrong with the script?

MichaIng, 2022-01-13:
> sed: Exec format error

That is an error with the `sed` executable itself. Can you try and paste the output of:
```sh
sed
```

XECDesign, 2022-01-13:
Yeah, seems more like a corrupted sd card than anything else.

JanWennrich, 2022-01-13:
> Can you try and paste the output of: `sed`

```
pi@raspberrypi:~ $ sed
-bash: /usr/bin/sed: cannot execute binary file: Exec format error
```

So I guess my SD card corrupted again? <.<

XECDesign, 2022-01-13:
Unless something overwrote your 'sed' or system libraries. Did you do anything interesting like add third party repos or run random mystery scripts?

JanWennrich, 2022-01-13:
No, I guess the SD card is messed up as my Raspberry had some other hiccups in the past days as well.  
I just reinstalled `sed` and then the script was executed flawlessly.

Sorry for any inconvenience and thanks for the fast help!

/close

closed 2022-01-13 by XECDesign