RPi-Distro/raspberrypi-sys-mods #62: Remove 010_pi-nopasswd

CodeTherapist, 2022-06-02:
In my humble opinion, the file [/etc/sudoers.d/010_pi-nopasswd](010_pi-nopasswd) should be removed or changed. Since the last release, the pi user is no longer a default user (ref: [raspberry-pi-bullseye-update-april-2022](https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/)).

Maybe I didn't got it, how it should work then I would appreciate any explanation.

pelwell, 2022-06-02:
I would imagine that the user creation process modifies that file to change the userid - why don't you try it?

XECDesign, 2022-06-06:
It's working as intended. General questions are better asked on the [forum](https://forums.raspberrypi.com/).

closed 2022-06-06 by XECDesign

CodeTherapist, 2022-06-07:
> I would imagine that the user creation process modifies that file to change the userid - why don't you try it?

Please don't assume I didn't try it - you don't know me.

> It's working as intended. General questions are better asked on the [forum](https://forums.raspberrypi.com/).

It works partially. I do have a private fork of the [pi-gen](https://github.com/RPi-Distro/pi-gen). In the past, setting the `FIRST_USER_NAME` and `FIRST_USER_PASS` with the `config` file was the way to go (what I did). The default user `pi` was recently removed, and the first-user-process changed drastically. Unfortunately, when something is set for `FIRST_USER_NAME` the replacement within the  `010_pi-nopasswd` doesn't work. That is why it was not working for me as expected.

XECDesign, 2022-06-07:
> Please don't assume I didn't try it - you don't know me.

It's because we don't know you that we can't assume that you did.

> Unfortunately, when something is set for `FIRST_USER_NAME` the replacement within the `010_pi-nopasswd` doesn't work.

That sounds like a legitimate issue with `pi-gen` that needs to be fixed, but it would've been there before the user rename scripts were added.

> @XECDesign What I did now, is a sub step within the staging steps to create the userconf in /boot. Wouldn't be that a good replacement to allow customized pi-gen repositories to still use `FIRST_USER_NAME/PASS`?

I think so, but some people weren't happy with that suggestion on the [pi-gen issue tracker](https://github.com/RPi-Distro/pi-gen/issues/614).

CodeTherapist, 2022-06-07:
@XECDesign Thanks for your response much appreciated 👍

gromain, 2022-06-10:
@CodeTherapist Sorry to bump this, but can you elaborate on your solution?
Do you create a userconf that respect what was set in `FIRST_USER_NAME/PASS`?

@XECDesign, as was explained in the issue you linked, the goal is to have a system already in place to respect `FIRST_USER_NAME/PASS` when they are set, otherwise, those options should just be removed from the config file and from the documentation.
If including something at the staging step that respect `FIRST_USER_NAME/PASS` works, then perfect.
I would be nice to avoid forcing people to maintain a lot of additional moving parts just to keep the previous behavior. As we established already, this change breaks backward compatibility in a lot of ways (and to be clear, I understand the security point of view).

CodeTherapist, 2022-07-13:
@gromain Yes, I did a additional step to put a userconfig into the boot with the info from `FIRST_USER_NAME/PASS`.
Unfortunately it seems still an issue - even with the new option `DISABLE_FIRST_BOOT_USER_RENAME=1`.