RPi-Distro/raspberrypi-sys-mods #63: New postinst flush code fails if vfat partition not found

theneofusion, 2022-09-08:
HW: RaspberryPi 4B
OS: Raspberry Pi OS Lite 64bit, otherwise up-to-date

raspberrypi-sys-mods (20220901) installs fine on systems running from SD and USB-drives but fails on my systems that network boot from a NFS share.

Error message:
`<snip>
Setting up libc-devtools (2.31-13+rpt2+rpi1+deb11u4) ...
Setting up raspberrypi-sys-mods (20220901) ...
Checking if boot partition needs 'flush' option...
dpkg: error processing package raspberrypi-sys-mods (--configure):
 installed raspberrypi-sys-mods package post-installation script subprocess returned error exit status 1
Setting up raspberrypi-net-mods (1.3.4) ...
Modified /etc/network/interfaces detected. Leaving unchanged and writing new file as interfaces.new.
Setting up libcamera-apps-lite (0~git20220830+1bf0cca-1) ...
Setting up libc6-dev:arm64 (2.31-13+rpt2+rpi1+deb11u4) ...
Setting up console-setup-linux (1.205+rpt1) ...
Setting up console-setup (1.205+rpt1) ...
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u4) ...
Errors were encountered while processing:
 raspberrypi-sys-mods
E: Sub-process /usr/bin/dpkg returned an error code (1)
> `

This is what my fstab looks like on the failing devices:
`$ cat /etc/fstab
proc            /proc           proc    defaults          0       0
#PARTUUID=6c872c37-01  /boot           vfat    defaults,flush    0       2
#PARTUUID=6c872c37-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
# PXE/TFTP boot
[obfuscated ip]:/srv/nfsroot/tftpboot/[obfuscated serialnr]  /boot           nfs    defaults          0       0
[obfuscated ip]:/srv/nfsroot/pxeroot/tank     /               nfs    defaults,noatime  0       0`

pdw-mb, 2022-09-09:
I think the problem is [this line](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/master/debian/raspberrypi-sys-mods.postinst#L14).  `findmnt` returns a non-zero exit code if not found, which causes the whole script to fail because `-e` is specified, so the code below to catch this gracefully is never reached.

I think adding `|| true` to the end of the `findmnt` command should fix it.

TeleMediaCC, 2022-09-09:
I have this error message too, when update raspberrypi-sys-mods. And I use TFTP boot too.

XECDesign, 2022-09-09:
That makes sense, thank you! I will upload the new build once it has been tested.

commit 7dd78cba (2022-09-09): Fix add_remove_flush_option behaviour when no boot partition is found

Fixes #63
commit 54722f8b (2022-09-09): Fix add_remove_flush_option behaviour when no boot partition is found

Fixes #63
closed 2022-09-09 by XECDesign