RPi-Distro/raspberrypi-sys-mods #64: [bullseye] FIRSTBOOT exits prematurely on NOOBS/PINN sd card

procount, 2022-09-12:
FIRSTBOOT does some partition checks, in particular that the root partition is the last partition and is on the same device as the boot partition. I assume these are to prevent needlessly resizing the root partition for NOOBS/PINN. However, this causes firstboot to exit completely and it does not do any of the other tasks such as ssh key generation or OS customisations. I think it should just prevent the partition resizing, but continue with the other tasks, or am I missing something?

(whilst OS customisations would not be applicable from rpi-imager, these files could be produced from other tools.)

XECDesign, 2022-09-14:
> I think it should just prevent the partition resizing

Yup, I'll fix that.

I would avoid doing anything with that customisations file until some kind of documentation about it goes out. Right now it's just for experimenting and is probably not a long-term solution. I suspect that it will eventually be replaced by cloud-init (if that gets to the point where it covers our use cases).

XECDesign, 2022-09-15:
I've just a potential fix, but I don't have a setup where I can test this easily. Would you be able to check whether it works as you expect?

https://github.com/RPi-Distro/raspberrypi-sys-mods/raw/master/usr/lib/raspberrypi-sys-mods/firstboot

procount, 2022-09-15:
My test involved updating the firstboot script on an installed Raspios that was not the last partition, and adding the init option back into cmdline.txt and rebooting it.
It did not attempt to resize the partition, but it did generate SSH keys.
I didn't test having partitions on different devices, but I reviewed the code and it looks ok now.
Many thanks.

XECDesign, 2022-09-16:
Thanks, we'll spin a new bug fix image in the near future, which should have this fix.

closed 2022-09-26 by XECDesign