RPi-Distro/raspberrypi-sys-mods #65: `init_config` python module named 'toml' is missing in Raspberry Pi OS lite 64bit

cernekj, 2022-09-14:
1) I have wrote a Raspberry Pi OS (64-bit) from <https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-09-07/2022-09-06-raspios-bullseye-arm64-lite.img.xz> to SDcard.
2) I have created `/boot/config.toml` configuration.
3) I started RPi and `/usr/lib/raspberrypi-sys-mods/firstboot` was executed.
4) There is an error in `/var/log/firstboot.log`:
```
  File "/usr/lib/raspberrypi-sys-mods/init_config", line 12, in <module>
    import toml
ModuleNotFoundError: No module named 'toml'
```
5) `toml` package is really missing in folder `/lib/python3.9/`.

So I suggest to install package `python3-toml` as part of the Raspios lite.

XECDesign, 2022-09-14:
That customisation method hasn't been mentioned and isn't an officially supported feature yet, so well spotted.

I'll add the fix, thanks.

XECDesign, 2022-09-16:
Should be fixed in the next set of images.

closed 2022-09-26 by XECDesign