RPi-Distro/raspberrypi-sys-mods #66: missing license

qyanu, 2022-09-19:
Hi,

is there a specific reason, where there is no LICENSE, COPYING or similar declaration (lone SPDX-License-Identifiers in each file probably do not suffice) stating under what license the content of this repository is published at github?
As i understand law, the current form of publication makes it open source but not free open source, meaning that any modifications i might make are in almost all cases *prohibited* to be shared with others.

So that's the question: why are the current state of affairs as they are?

best regards,
Max

MichaIng, 2022-09-19:
https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/master/debian/copyright

Could be added to the repositories root indeed, a way that it is shown at the right sidebar of the GitHub UI.

qyanu, 2022-09-19:
I kind of looked into that directory, but apparently not with a very high amount of rigor.

closed 2022-09-19 by qyanu