RPi-Distro/raspberrypi-sys-mods #74: Abstract boot partition path

dsx724, 2023-05-16:
Can you abstract the hard-coded boot path /boot to a variable so that it can be changed in one place instead of half a dozen?

I'm using GRUB EFI with some other OSes and this is requiring a lot of hacking.

XECDesign, 2023-05-18:
In the process of doing something slightly related. Out of curiosity, what would the path you'd use be?

dsx724, 2023-05-23:
Having it align with EFI (/boot/efi) would be great. But I know this is unrealistic due to the amount of hacks people have made for /boot so having it abstracted and changed in one place is better than making unreliable sed's all over the place.

ragazenta, 2023-10-16:
Hi, have these changes been pushed somewhere on private repo? I see that bookworm version already have these changes. Thanks.

XECDesign, 2023-10-16:
Yes sorry, forgot to push the new branch. Should be there now.

closed 2023-10-16 by XECDesign