RPi-Distro/raspberrypi-sys-mods #84: Serial port symlinks broken in bookworm.

metachip, 2023-10-25:
There appears to be a bug in the assignment of the primary and secondary serial ports in bookworm.

It looks like a problem in:

raspberrypi-sys-mods: /etc/udev/rules.d/99-com.rules

When bluetooth is disabled in config.txt as:

dtoverlay=disable-bt

I would expect that

/dev/serial0 -> ttyAMA0
/dev/serial1 -> ttyS0

and this is what bullseye does, correctly, but instead with bookworm we see:

/dev/serial1 -> ttyAMA0

and /dev/serial0 is not symlinked at all.

This is contrary to the advice here:

https://www.raspberrypi.com/documentation/computers/configuration.html#primary-and-secondary-uart

and looks like a bug in bookworm :)

XECDesign, 2023-10-25:
@pelwell, is this something that has been or will be fixed with newer firmware/kernel or just expected behaviour?

pelwell, 2023-10-25:
It's definitely not expected behaviour, and it may have been fixed with the latest firmware (coming to an apt upgrade soon) - knowing which Pi it was would help.

metachip, 2023-10-25:
# Serial Device Alias Fault Diagnosis

The problem is the line:

readlink /sys/class/tty/ttyS[0-9]*/device/of_node | sed 's/base/:/' | cut -d: -f2

in

/etc/udev/rules.d/99-com.rules

On Pi5 it works.

On Pi4 it does not.

The problem is the format of the descriptor for ttyS0 is different.

On Pi5:

$ echo `cat /proc/device-tree/model`
Raspberry Pi 5 Model B Rev 1.0
$ tail -2 /boot/firmware/config.txt
enable_uart=1
dtoverlay=disable-bt
$ readlink /sys/class/tty/ttyS0/device/of_node
../../../../firmware/devicetree/base/soc/serial@7d50c000

On Pi4:

$ echo `cat /proc/device-tree/model`
Raspberry Pi 4 Model B Rev 1.4
$ tail -2 /boot/firmware/config.txt
enable_uart=1
dtoverlay=disable-bt
$ readlink /sys/class/tty/ttyS0/device/of_node
$ ls /sys/class/tty/ttyS0/device/
driver driver_override modalias power subsystem tty uevent

Bottom line:

/etc/udev/rules.d/99-com.rules

requires modification to work on models other than Pi5. The test on  **of_node**  fails on Pi4 because it does not exist.

pelwell, 2023-10-26:
The real problem is that the use of the bluetooth alias to determine which UART is which should be conditional on the existence of the console alias, not vice-versa:
```diff
diff --git a/etc.armhf/udev/rules.d/99-com.rules b/etc.armhf/udev/rules.d/99-com.rules
index 1931d3d..7ad5f38 100644
--- a/etc.armhf/udev/rules.d/99-com.rules
+++ b/etc.armhf/udev/rules.d/99-com.rules
@@ -17,11 +17,11 @@ SUBSYSTEM=="pwm", ACTION!="remove", PROGRAM="/bin/sh -c 'chgrp -R gpio /sys%p &&
 KERNEL=="ttyAMA[0-9]*|ttyS[0-9]*", PROGRAM="/bin/sh -c '\
         ALIASES=/proc/device-tree/aliases; \
         TTYNODE=$$(readlink /sys/class/tty/%k/device/of_node | sed 's/base/:/' | cut -d: -f2); \
-        if [ -e $$ALIASES/bluetooth ] && [ $$TTYNODE/bluetooth = $$(strings $$ALIASES/bluetooth) ]; then \
-            echo 1; \
-        elif [ -e $$ALIASES/console ]; then \
+        if [ -e $$ALIASES/console ]; then \
             if [ $$TTYNODE = $$(strings $$ALIASES/console) ]; then \
-                echo 0;\
+                echo 0; \
+            elif [ -e $$ALIASES/bluetooth ] && [ $$TTYNODE/bluetooth = $$(strings $$ALIASES/bluetooth) ]; then \
+                echo 1; \
             else \
                 exit 1; \
             fi \
```
Doing so with `dtoverlay=disable-bt` results in:
```
/dev/serial0 -> ttyAMA0
```
which is arguably more correct than:
```
/dev/serial0 -> ttyAMA0
/dev/serial1 -> ttyS0
```
since /dev/ttyS0 doesn't actually work (try it - the device node is disabled, and yet somehow the driver is creating a /dev/ entry for it, perhaps because of `8250.nr_uarts=1` on the command line).

pelwell, 2023-10-27:
There's now a PR with this change: #85

hasenradball, 2024-12-20:
@pelwell Hi,

I have an actual bookworm on Pi 2, and Iam not able to get the uart0 work on GP14/15.
I this the same issue?

See [Serial not working](https://forums.raspberrypi.com/viewtopic.php?t=375382#p2245844)

pelwell, 2024-12-20:
It seems unlikely. I'm going to largely ignore the contents of the thread and ask you to state (or restate) here, together, the output from the following commands (or the answers, if they are questions):
1. `ls -l /dev/ttyS0 /dev/ttyAMA0 /dev/serial*`
2. `pinctrl | grep XD`
3. `cat /proc/cmdline`
4. If `/dev/ttyS0` appeared in the output of 1, `sudo stty -F /dev/ttyS0 -a`
5. If `/dev/ttyAMA0` appeared in the output of 1, `sudo stty -F /dev/ttyAMA0 -a`
6. How are you trying to use the serial port?

hasenradball, 2024-12-20:
hi @pelwell ,

thanks for answering.
Here my results:

1) `ls -l /dev/ttyS0 /dev/ttyAMA0 /dev/serial*`
```
ls: Zugriff auf '/dev/ttyS0' nicht m▒glich: Datei oder Verzeichnis nicht gefunden
lrwxrwxrwx 1 root root          7 20. Dez 22:27 /dev/serial0 -> ttyAMA0
crw-rw---- 1 root dialout 204, 64 20. Dez 22:27 /dev/ttyAMA0
```
2) `pinctrl | grep XD`
```
14: a0    -- | hi // GPIO14 = TXD0
15: a0    -- | hi // GPIO15 = RXD0
```
3) `cat /proc/cmdline`
```
coherent_pool=1M snd_bcm2835.enable_headphones=0 cgroup_disable=memory snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  overlayroot=tmpfs root=PARTUUID=7b325855-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash cfg80211.ieee80211_regdom=DE
```
5) `sudo stty -F /dev/ttyAMA0 -a`
```
speed 115200 baud; rows 24; columns 80; line = 0;
intr = ^C; quit = ^\; erase = ^?; kill = ^U; eof = ^D; eol = <undef>; eol2 = <undef>; swtch = <undef>; start = ^Q; stop = ^S; susp = ^Z; rprnt = ^R; werase = ^W; lnext = ^V; discard = ^O; min = 1; time = 0;
-parenb -parodd -cmspar cs8 hupcl -cstopb cread clocal -crtscts
-ignbrk -brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl -ixon -ixoff -iuclc -ixany -imaxbel iutf8
opost -olcuc -ocrnl onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
-isig -icanon -iexten -echo -echoe -echok -echonl -noflsh -xcase -tostop -echoprt -echoctl -echoke -flusho -extproc
```
6) 
I use putty and tried to use toe following:

- `/dev/ttyAMA0`
- `/dev/serial0`
- `/dev/ttyS0`

I also used an loopback connector when starting putty, nothing appears on the terminal window.

pelwell, 2024-12-20:
That's good - almost everything is as I would expect. If you want to be able to log in from another computer over the UART then you will need to add 'console=serial0,115200' to `/boot/firmware/cmdline.txt` (keeping it all in a single line), then reboot.

hasenradball, 2024-12-20:
Hi no I want to debug my ESP8266 via UART0.
But I am wondering why I get nothing on the output?
hmmm

hasenradball, 2024-12-22:
Hi @pelwell,
I tried a second try.

Use a pi 2 do an fresh image download of bookworm 32-bit.

After the system runs I installed putty.
Setup the serial
/dev/ttyAMA0.

But if I want to open then the terminal it seems putty crashes.
Putty closes window - done.

My impression is, that in bookworm somethings going totally wrong with Serial.
This is seen in also many issues seen in pi4 and pi5.

With the many releases before there was never an  such a long lasting issue with serial and putty.

pelwell, 2024-12-23:
I don't think there is a problem with UARTs in RPiOS, either generally or specifically on Pi 2. If there were, you wouldn't be the only person complaining.

Here are a number of tests you can perform to try and locate the issue:
1. Disconnect the serial cable. What do the following command report? Perhaps just copy and paste whole dialog:
```
$ pinctrl 14-15
$ pinctrl 14-15 ip pn
$ pinctrl 14-15
$ pinctrl 14-15 op dh
$ pinctrl 14-15
$ pinctrl 14-15 op dl
$ pinctrl 14-15
$ pinctrl 14-15 ip pd
```
2. With the loopback between GPIOs 14 & 15 (pins 8 & 10):
```
$ pinctrl 14-15
$ pinctrl 14 op dh
$ pinctrl 14-15
$ pinctrl 14 op dl
$ pinctrl 14-15
$ pinctrl 14 ip
$ pinctrl 15 op dh
$ pinctrl 14-15
$ pinctrl 15 op dl
$ pinctrl 14-15
$ pinctrl 15 ip pu
$ pinctrl 14-15 a0
$ pinctrl poll 14-15 &
$ echo hello > /dev/ttyAMA0
$ killall pinctrl
```

hasenradball, 2024-12-23:
Hi thank you for watching,

I hope I am the only one who complains about that.
It is totally wired, there was no issue the last ten years about that. But on pi2 there should be no issue because pi2 has no overlayed bluetooth and stuff.

I will check your commands, thank you.

In meanwhile let the jingle bell.

Happy christmas

hasenradball, 2024-12-23:
Totally wired was today with a fresh bookworm installation and fresh putty installation.
Opened in putty the
 `/dev/ttyAMA0`
And putty crashed (window closed and did not popup again)…

hasenradball, 2024-12-23:
The actual issue I have is.
If you start putty with `/dev/ttyAMA0` on pi 2 putty does not open a terminal window.

hasenradball, 2024-12-23:
Where to place this issue?

hasenradball, 2024-12-25:
> I don't think there is a problem with UARTs in RPiOS, either generally or specifically on Pi 2. If there were, you wouldn't be the only person complaining.
>
> Here are a number of tests you can [...]

Here you will find the result:

```
uname -a

Linux pi2 6.6.62+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.62-1+rpt1 (2024-11-25) armv7l GNU/Linux

/etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Start pinctrl test
pinctrl 14-15
14: a0    -- | hi // GPIO14 = TXD0
15: a0    -- | hi // GPIO15 = RXD0

pinctrl 14-15 ip pn
14: ip    -- | lo // GPIO14 = input
15: ip    -- | lo // GPIO15 = input

pinctrl 14-15 op dh
14: op -- -- | hi // GPIO14 = output
15: op -- -- | hi // GPIO15 = output

pinctrl 14-15 op dl
14: op -- -- | lo // GPIO14 = output
15: op -- -- | lo // GPIO15 = output

pinctrl 14-15 ip pd

 ==> set loopback connection
14: ip    -- | lo // GPIO14 = input
15: ip    -- | lo // GPIO15 = input

pinctrl 14 op dh
14: op -- -- | hi // GPIO14 = output
15: ip    -- | hi // GPIO15 = input

pinctrl 14 op dl
14: op -- -- | lo // GPIO14 = output
15: ip    -- | lo // GPIO15 = input

pinctrl 14 ip

pinctrl 15 op dh
14: ip    -- | hi // GPIO14 = input
15: op -- -- | hi // GPIO15 = output

pinctrl 15 op dl
14: ip    -- | lo // GPIO14 = input
15: op -- -- | lo // GPIO15 = output

pinctrl 15 ip pu

pinctrl 14-15 a0

pinctrl poll 14-15 &

echo hello > /dev/ttyAMA0
14: lo // GPIO14
15: lo // GPIO15
+2us
14: hi // GPIO14
15: hi // GPIO15
+52us
14: lo // GPIO14
15: lo // GPIO15
+73us
14: hi // GPIO14
15: hi // GPIO15
+76us
14: lo // GPIO14
15: lo // GPIO15
+389us
14: hi // GPIO14
15: hi // GPIO15

killall pinctrl
Test finished

```

pelwell, 2025-01-01:
That output all looks correct - you even see the output changing when characters are sent. I see there is one case the test misses, which is a short between the GPIOs (although manufacturing tests would have picked this up). You could try to manipulate the two pins independently - making one an output driving in the two directions and one an input, with the loopback removed, confirming that the input isn't changed by the output.

I suggest you run this test program (https://forums.raspberrypi.com/viewtopic.php?p=1679003#p1679003) with the loopback fitted - I've not tried it myself but it should work, showing the letters A to Z.

hasenradball, 2025-01-01:
@pelwell 
Hi HNY2025!

I think I have both tests done - independently and shortend between 14/15.

See here:

```

14: ip    -- | lo // GPIO14 = input
15: ip    -- | lo // GPIO15 = input

==> set loopback connection
pinctrl 14 op dh
14: op -- -- | hi // GPIO14 = output
15: ip    -- | hi // GPIO15 = input

pinctrl 14 op dl
14: op -- -- | lo // GPIO14 = output
15: ip    -- | lo // GPIO15 = input

```
Are this the test you missed?

hasenradball, 2025-01-01:
@pelwell 
But beeing honest if all port checks are ok.
Could it be a acccessrights issue?

And how could you explain that on bullseye it works directly, and on bookwoorm there is no chance.

For me it seems to be a generic issue on book worm maybee related to pi 2.

Could you make a short test please on pi 2 for this?

Many weeks ago I also reported an serial on pi 5, where all developpers said there is no issue.

But at the end there was an issue!

pelwell, 2025-01-01:
> Are this the test you missed?

Those tests are with the loopback link between the pins. I'm trying to show that without the link the two pins behave independently. Can you see a line in your testing output where GPIOs 14 and 15 have different levels?

I won't be able to test on a Pi 2 until next week.

hasenradball, 2025-01-01:
@pelwell 
understood, will do the tests.

Would be great if you could mak axshortctestcon pi 2.

hasenradball, 2025-01-01:
@pelwell 
here the pin independency test:

```
uname -a

Linux pi2-garage 6.6.62+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.62-1+rpt1 (2024-11-25) armv7l GNU/Linux

/etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Start pinCtrlTest2
Check independency of Pins

pinctrl 14-15
14: ip    -- | lo // GPIO14 = input
15: ip    -- | lo // GPIO15 = input

pinctrl 14 op dh
pinctrl 15 ip pd
14: op -- -- | hi // GPIO14 = output
15: ip    -- | lo // GPIO15 = input

pinctrl 14 op dl
pinctrl 15 ip pu
14: op -- -- | lo // GPIO14 = output
15: ip    -- | hi // GPIO15 = input

pinctrl 14 ip pd
pinctrl 15 op dh
14: ip    -- | lo // GPIO14 = input
15: op -- -- | hi // GPIO15 = output

pinctrl 14 ip pu
pinctrl 15 op dl
14: ip    -- | hi // GPIO14 = input
15: op -- -- | lo // GPIO15 = output

pinctrl 14 ip pd
pinctrl 15 ip pd
14: ip    -- | lo // GPIO14 = input
15: ip    -- | lo // GPIO15 = input
Test finished

```

hasenradball, 2025-01-01:
@pelwell 
And here the loopback test:

```

frank@pi2-garage:~/Documents/python $ nano serialLoopbackTest.py
frank@pi2-garage:~/Documents/python $ python serialLoopbackTest.py /dev/serial0 115200
Loopback Test on /dev/serial0 at 115200 baud
^C
frank@pi2-garage:~/Documents/python $
frank@pi2-garage:~/Documents/python $ python serialLoopbackTest.py /dev/ttyAMA0 115200
Loopback Test on /dev/ttyAMA0 at 115200 baud
^C
frank@pi2-garage:~/Documents/python $

```

No output printed I have aborted manually.

pelwell, 2025-01-01:
And that is with the GPIOs connected together?

hasenradball, 2025-01-01:
> And that is with the GPIOs connected together?

Yes exactly.
With the pins connected.

hasenradball, 2025-01-07:
@pelwell 
Hi thanks for your feedback.
Habe you used a light os or an os with GUI?

pelwell, 2025-01-07:
It's the GUI version. I can try the Lite if necessary.

hasenradball, 2025-01-07:
And the latest bookworm?

The issue I have was latest bookworm with GUI.
Will try a last try in a few hours.
I tried on 2 pi2.

pelwell, 2025-01-07:
Absolute latest bookworm downloaded from RPi Imager.

hasenradball, 2025-01-07:
Ok 
tested with OS lite on the pi2.

The serial loopback test works here with os lite.
Now i switch back to the other sd card with the installed os (non lite) and try the same.

hasenradball, 2025-01-07:
@pelwell 
Ok I can confirm that  the `uarttest.py` works for me on both versions (lite, and non-lite).

But and thats the point I complained at beginning.

After I successfully checked the uarttest on the non-lite os , I opened putty with the placed loopback connection.
But putty does not work with serial (`/dev/ttyAMA0`).
After I tried to open putty in serial it seams putty chrashed.

Can you confirm this, please on your machine?

Thank you so much.

hasenradball, 2025-01-08:
@pelwell 
have you changed something in the uarttest?
because last time I tried there was no Output

pelwell, 2025-01-08:
No, the script hasn't changed - my gist is still as it was, and a verbatim copy of the one in the forum post.

I just install putty and ran it. Connecting to the UART causes it to exit, but so does trying to use an SSH connection. If you launch putty from the command line (just run `putty`) you see a few Gtk complaints, then when you try to open the session it say `unable to load font "server:fixed"`.

Open putty again, but this time navigate to the Window/Fonts tab; notice how the current font for ordinary text is "server:fixed". Click the Change... button and choose any font from the list - I like FreeMono. Now try the UART again - it works for me.

hasenradball, 2025-01-08:
Ok understood what to test.
will do an give feedback.
thank you.

hasenradball, 2025-01-08:
@pelwell 
ok I checked it was indeed the font issue from putty.
when I changed the font everything works as expected.

Sorry for the inconveniences.
And thank you so much for the help.

pelwell, 2025-01-08:
We're discussing if there is a way of fixing this in the OS.

closed 2025-01-08 by pelwell

hasenradball, 2025-01-09:
@pelwell  One additional question.
What do you think is the root cause?
- no error meassage
- wrong default font in putty

Or both?
Would it not also an additional option to set default font in putty to „Monospace“?

pelwell, 2025-01-10:
The cause of the error is the missing font, or more correctly that PuTTY running under Wayland can't find the X11 font. There is another way to solve the problem, which is to add the following to your environment, forcing PuTTY to run in X11 mode:
```
GDK_BACKEND=x11
```
The error message was generated, just not visible unless you run PuTTY from the command line (it may have been logged somewhere...).

lurch, 2025-01-17:
A quick search shows that this also affects other distributions when running Putty under Wayland: https://www.google.com/search?q=putty+linux+missing+font+wayland

And ironically one of those search results points to our own forums! 😆  https://forums.raspberrypi.com/viewtopic.php?t=378681

lurch, 2025-01-17:
> We're discussing if there is a way of fixing this in the OS.

https://gitlab.archlinux.org/archlinux/packaging/packages/putty/-/issues/2#note_202404 suggests setting the `-D NOT_X_WINDOWS` CFLAG.