RPi-Distro/raspberrypi-sys-mods #90: Custom script during firstboot

knachte, 2024-05-14:
I've changed /usr/lib/raspberrypi-sys-mods/firstboot to include a check for a file called 'custom' in the same location. If it is present it executes that file. Would you be ok with a pull request for something like that? This would enable easier inclusion of custom logic during mass production of devices, which need to make calls to api's for configuration.

This could be done with a separate service, but then the reboot from the firstboot script might happen before  all the api calls have returned and cause the custom logic to fail.

XECDesign, 2024-05-15:
Currently, something like this could be accomplished using the same mechanism rpi-imager uses to customise images. I don't want to add yet another custom non-standard thing to support. No matter what we do, somebody will come along and say "yeah, but I need the network to be available, and can it also set up static IPs on my network interfaces for me and build a kernel module?"

We really need to add cloud-init support  (apologies that it's taking this long). The advantage of cloud-init would be that it's the de-facto standard for doing this kind of thing with a much larger set of features.

knachte, 2024-05-15:
I wasn't aware that something like cloud-init was coming. I'll use my custom stuff for now and switch over when that is available. thanks!

closed 2024-05-15 by knachte