RPi-Distro/raspberrypi-sys-mods #94: First boot customization?

inferiorhumanorgans, 2025-02-28:
Per this comment it sounds like cloud-init support was the end goal.  However it looks like that's been abandoned and customizing the first boot is no longer (officially) supported?  Are there any plans to at least document config.toml which has lots of paper cuts that aren't documented anywhere?

https://github.com/RPi-Distro/raspberrypi-sys-mods/issues/82#issuecomment-1779531470

MichaIng, 2025-02-28:
With the Raspberry Pi Imager, you can pre-configure the image: https://www.raspberrypi.com/documentation/computers/getting-started.html#raspberry-pi-imager

inferiorhumanorgans, 2025-02-28:
Sorry, I should clarify: in a repeatable and scriptable manner.  Before bookworm and the systemd disaster this was possible in a supported manner.

MichaIng, 2025-02-28:
Ah right, that is somewhat lacking. `cloud-init` AFAIK is not abandoned, is still the final goal, but no one started working on it yet/no ETA.

On first boot, `/usr/lib/raspberrypi-sys-mods/firstboot` is executed. So that one can be used for customization. It also handles the `custom.toml`, which is finally done by the Python script `/usr/lib/raspberrypi-sys-mods/init_config`. So from that script you can derive available settings. Like this should create a user with password and enable SSH with password authentication:
```toml
[user]
name = "michaing"
password = "some_plain_password"
is_encrypted = false

[ssh]
enabled = true
password_authentication = true
```
The password can be also given encrypted, but would need to check the algorithm. An SSH key can be provided, but I am currently not sure how to define it correctly. Probably like this as array:
```toml
authorized_keys = [ "ssh-ed25519 XXXXXXXXXXXXXXXX comment" ]
```

XECDesign, 2026-09-09:
Closing as stale.

cloud-init turned out to be a dead end. Customisation will be done by https://github.com/raspberrypi/rpi-preseed in the future.

closed 2026-09-09 by XECDesign