RPi-Distro/raspberrypi-sys-mods #96: udev rule not assigning ttyAMA serial links

drewmnoel, 2025-04-13:
After #95 was merged, ttyAMA0 doesn't get a serial dev entry if UART is enabled (i.e., it doesn't get `serial1` nor clobber `serial0`:

```
$ ls -l /dev/serial*
lrwxrwxrwx 1 root root 5 Apr 12 23:19 /dev/serial0 -> ttyS0

$ ls -l /dev/ttyA*
crw-rw---- 1 root dialout 204, 64 Apr 12 23:19 /dev/ttyAMA0
```

I'm not positive how to debug udev but I believe the problem is just a typo in the final sed:

```diff
@@ -13,7 +13,7 @@

 KERNEL=="ttyAMA[0-9]*|ttyS[0-9]*", PROGRAM="/bin/sh -c '\
         ALIASES=/proc/device-tree/aliases; \
-        TTYNODE=$$(readlink /sys/class/tty/%k/device | sed 's/.*\\//\\/soc\\/serial@/' | sed 's/\\..*//' | sed 's/@f/@7/' | sed 's/@3f/@fe' ); \
+        TTYNODE=$$(readlink /sys/class/tty/%k/device | sed 's/.*\\//\\/soc\\/serial@/' | sed 's/\\..*//' | sed 's/@f/@7/' | sed 's/@3f/@fe/' ); \
         if [ -e $$ALIASES/console ]; then \
             if [ $$TTYNODE = $$(strings $$ALIASES/console) ]; then \
                 echo 0; \
```

Testing locally, udev now correctly gives the numbers with that change:

```
$ ls -l /dev/serial*
lrwxrwxrwx 1 root root 5 Apr 12 23:14 /dev/serial0 -> ttyS0
lrwxrwxrwx 1 root root 7 Apr 12 23:14 /dev/serial1 -> ttyAMA0
```

pelwell, 2025-04-14:
Yes - it looks correct, and it works for me. I'll submit a PR.

pelwell, 2025-04-14:
See https://github.com/RPi-Distro/raspberrypi-sys-mods/pull/97.

closed 2025-04-16 by pelwell