RPi-Distro/raspberrypi-sys-mods #102: Pre-configured WiFi via Raspberry Pi Imager does not take effect on trixie.

jvonau, 2025-08-30:
Raspberry Pi Imager v1.9.6 creates the firstrun.sh referencing /usr/lib/raspberrypi-sys-mods/imager_custom which in raspberrypi-sys-mods_20250605~bookworm_arm64.deb that file is present 

[image]

However the [trixie nightlies](https://downloads.raspberrypi.com/nightlies/) installs raspberrypi-sys-mods_20250630_arm64.deb (check the info file) but that file is not part of the of the later deb

[image]

This results in having a wpa_supplicant.conf configured that does not work with NetworkManager and not the preconfigured.nmconnection that would be provided by imager_custom.

excerpt from firstboot.sh
>if [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then
   /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan 'SHAW-EE5B' 'redacted' 'CA'
else
cat >/etc/wpa_supplicant/wpa_supplicant.conf <<'WPAEOF'
country=CA
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
ap_scan=1

>update_config=1
network={
	ssid="SHAW-EE5B"
	psk=redacted
}

>WPAEOF
   chmod 600 /etc/wpa_supplicant/wpa_supplicant.conf
   rfkill unblock wifi
   for filename in /var/lib/systemd/rfkill/*:wlan ; do
       echo 0 > $filename
   done
fi

I was looking to have working WiFi on the first boot but did not.

holta, 2025-08-30:
Should this be mentioned within https://github.com/raspberrypi/rpi-imager/issues ?

XECDesign, 2025-09-01:
That's expected. We haven't released a Trixie image yet. When we do, it will either support cloud-init or we will roll back the change that removed `imager_custom`.

closed 2025-09-01 by XECDesign